bambulab/BambuStudio #3496: P1P gets stuck in "Downloading" state if sliced object has any "Pause" in G-code

Zaigon850, 2024-01-28:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1P

### How to reproduce

1. Load .3mf file in Bambu Studio
2. Slice with standard settings
3. Add in "Pause" command on any level, slice again
4. Send to printer
5. Bambu Studio sends file
6. Printer starts downloading file, bed starts to heat up

### Actual results

When Printer reaches "Downloading...(100%)" nothing more happens.
[image]
[image]

### Expected results

The Printer should Extract print job and commence Bed Leveling procedure.

### Project file & Debug log uploads

[Skylt Diskmaskin.zip](https://github.com/bambulab/BambuStudio/files/14075656/Skylt.Diskmaskin.zip)
[debug_network_Sun_Jan_28_11_12_10.log.enc.zip](https://github.com/bambulab/BambuStudio/files/14075672/debug_network_Sun_Jan_28_11_12_10.log.enc.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2024-02-04:
Hi. @Zaigon850 
Is this a reproducible issue or did it occur only once? I have sent the printer's log to the developer, and I believe there will be a solution soon.

walterwongbbl, 2024-02-04:
@Zaigon850 
Can you provide your printer logs? They should be located on the SD card.

Zaigon850, 2024-02-05:
Hi, sorry for late answer!

I've formatted the SD card (via the printer's interface), because the printer said it wasn't enough space on it. That didn't solve the issue though.

I gave up and I haven't tested printing any sliced models with pauses in the g-code since then. It works with regular sliced models, they're printing fine.

Regards
Tim Appelqvist

walterwongbbl, 2024-02-21:
@Zaigon850 
Thank you for your suggestion. We will consider supporting these in the future.

closed 2024-02-21 by walterwongbbl