bambulab/BambuStudio #3497: Need interval go back to clean up the nozzle for large printing.

swdjojo, 2024-01-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

When I'm printing a large object, ex. 12 hours's printing. There was a small amount of leakage filament on side of nozzle. Normally this is no problem. But once the leakage adheres to the printed object, it will cause the nozzle to hit the object, causing the printed object to fall off and fail.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

If there is a feature setting can interval by layer or print time, let nozzle can go back to clean up, that will helpful to prevent this situation.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

If I see that situation, I will pause print, then resume print, it will go back when pause, then clean up when resume.

**Additional context**
Add any other context or screenshots about the feature request here.

kvnper, 2024-02-29:
I've also experienced this issue

victornpb, 2024-05-20:
If you see this happening you can pause and unpause the printer from the app, if i remember it wipes when you unpause