bambulab/BambuStudio #3498: All Project Disappeared

tk4108, 2024-01-28:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 14.2.1

### Additional system information

M1; 16GB

### Printer

X1C

### How to reproduce

Forced quit after the application stopped responding.

### Actual results

Application crashed and all projects have been lost. The icons for the projects are still there, but it says "This project is no longer available."

### Expected results

My projects should still be on the project tab.

### Project file & Debug log uploads

[Archive.zip](https://github.com/bambulab/BambuStudio/files/14076140/Archive.zip)
[HELMET.3mf.zip](https://github.com/bambulab/BambuStudio/files/14076158/HELMET.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2024-01-29:
Could you provider a picture too show the problem?
Please help check if your 3mf file is still there? If any files have been deleted, please provide a file name for tracing.

tk4108, 2024-02-03:
I'm sorry. Deleted the projects and started over. The project files were not moved. I had to start a new project reopen the file for everything on my home screen.

bambu123, 2024-02-06:
> I'm sorry. Deleted the projects and started over. The project files were not moved. I had to start a new project reopen the file for everything on my home screen.
> […](#)

So, do you still have the original problem? If so, please provide some desktop screenshots. For the Crash problem, I need the DMP file under the log directory for analysis. Please help send it to me.

tk4108, 2024-02-09:
Hello,

The problem has not happened again with the projects I've opened. Unfortunately, I don't see a dmp file in the log directory. Sorry!

Cheers!
Matthew

bambu123, 2024-02-19:
I think we can close this issue now. You can reopen it at any time if you encounter the problem again.

closed 2024-02-19 by bambu123