bambulab/BambuStudio #3503: Slicing Issues/Adding Random dots to places that shouldn't be there

LaZyPrintz, 2024-01-29:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: Intel Core i7 13700k, Ram: 64GB, SSD: 3TB: Motherboard: Asus Prime z790, GPU: RTX 4070, PSU: 850W Gold

### Printer

Bambu Lab P1S

### How to reproduce

1. Go to Bambu Studios and open a lightbox file
2.  Confirm settings and slice said lightbox
3. Check layer 2 of the toolhead path for random dots of a different color than what is there
4. Print the lightbox and you end up getting dots of different colors where there shouldn't be any dots

### Actual results

The result is that now my lightboxes have random dots of another color behind the right color so when light shines through, you can see a random dot that's not supposed to be there.

In the picture I'm attaching, you'll see those dots as yellow dots. Most are near the edge of other colors. The main one on this one that messes up is next to the letters OU. I'm also attaching a picture of it colored in black but theres no yellow or black near that area.
[image]

[image]

### Expected results

It should slice normally where the colors that I colored it in the paint option shows up and doesn't add random dots. It's done this on the last 3 models I've designed and the 1st and 3rd layers on up do great but its just that second layer and it does it with whatever color is in filament slot 1.

### Project file & Debug log uploads

[OU Helmet Lamp (Ready to Print).zip](https://github.com/bambulab/BambuStudio/files/14090490/OU.Helmet.Lamp.Ready.to.Print.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-01-30:
[image]

Is the file you uploaded the same as the file you used? I can't reproduce your problem here.

zhimin-zeng-bambulab, 2024-08-20:
close as cannot reproduce, and no response for a long time.

closed 2024-08-20 by zhimin-zeng-bambulab