bambulab/BambuStudio #3508: boolean mesh does not work correctly

Gentle-Camper, 2024-01-30:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu LAB P1S

### How to reproduce

I create objects in the Bambu LAB Studio that I want to combine using the "Boolean Mesh" function. This also usually works with the first objects (including negative objects). But as soon as I try to use this function more often, I get an error message.

### Actual results

Error:

A Boolean operation cannot be performed on model meshes.
Only positive parts are exported.

### Expected results

All Boolean operations must be carried out because the printer converts the object correctly and the printer also prints the object correctly.

### Project file & Debug log uploads

[Haken für Abwasserschlauch.zip](https://github.com/bambulab/BambuStudio/files/14099967/Haken.fur.Abwasserschlauch.zip)
[debug_Mon_Jan_29_13_38_30_7668.log.zip](https://github.com/bambulab/BambuStudio/files/14099985/debug_Mon_Jan_29_13_38_30_7668.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2024-01-31:
The 3mf file you uploaded has only one part and can't do boolean. Did you upload the wrong file?

Gentle-Camper, 2024-01-31:
[Haken für Abwasserschlauch.zip](https://github.com/bambulab/BambuStudio/files/14107743/Haken.fur.Abwasserschlauch.zip)

Sorry, it was the wrong file.
Now it should be the right one.

tangketan, 2024-02-05:
You can use MeshLab to view the self-intersected faces:
[image]

Gentle-Camper, 2024-02-05:
Thanks for the help, I'll try it out.

Gentle-Camper, 2024-02-05:
I tried it, but that's not the problem.
I have a positive object and then want to subtract a negative object. Then combine these with a Boolean mesh. This works sometimes and not always. In the image you sent me, the negative object was simply left out.

tangketan, 2024-03-04:
@Gentle-Camper Mesh boolean has stronger requirements of object meshes than slicing. If one of your objects, either positive or negative, has self-intersection, boolean can't be done. In this case, the positive object has self-intersection - the fonts are just attached to the cylinder but not fused.
Self-intersection is difficult to resolve. I'm not giving a solution - there is nothing we can do now. You'll need to re-design your object mesh.

douglasg14b, 2025-05-19:
bad bot