bambulab/BambuStudio #3510: Bambu Studio Grey Screens after updating MacOS to 14.3

zenourn, 2024-01-30:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.3

### Additional system information

Apple M1 Max 64 GB

### Printer

Bambu Lab P1S

### How to reproduce

1. Open Application
2. Select New Project or Open Project

### Actual results

[image]

### Expected results

Normal Bambu Studio Project Window

### Project file & Debug log uploads

[Logs.zip](https://github.com/bambulab/BambuStudio/files/14104765/Logs.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2024-01-31:
from log, it is related with opengl
2024-01-31 10:47:34.158268[Thread 0x00000001e1571c40]:10:47:34鈥痑m: Couldn't create OpenGL context

@zenourn
could you help to check whether [prusaslicer](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.1) working or not?
and is there any way to check whether opengl still available on your system or not

zenourn, 2024-01-31:
Thanks - prusaslicer had the same issue, it came up with the error message that it Couldn't create an OpenGL context. I replaced all OpenGL system files and now it is working again. Why the OS update broke this is unclear, although OpenGL has been officially deprecated by Apple and I'm not sure how much longer OpenGL is going to be supported. Closing this as a MacOS rather than Bambu Studio problem.

closed 2024-01-31 by zenourn

lanewei120, 2024-01-31:
@zenourn 
Glad to hear that it works

could you also help to show us which files have you replaced?
it can help other customers with the same issue in the future

zenourn, 2024-01-31:
I copied the OpenGL files from another machine running 14.3 in /System/Library/Frameworks/OpenGL.framework.  It should technically be very difficult for these files to become corrupted, so I'm slightly suspicious about this whole issue.

lanewei120, 2024-02-01:
@zenourn 
thanks very much~

Andrew-Pynch, 2024-06-22:
@zenourn @lanewei120 

would either of you be willing to share you opengl files? I am also having this issue for the same reason, but I dont have a different mac to get clean files off of

lanewei120, 2024-06-25:
> @zenourn @lanewei120
> 
> would either of you be willing to share you opengl files? I am also having this issue for the same reason, but I dont have a different mac to get clean files off of

@Andrew-Pynch whay is your macos version
and also what is the cpu