bambulab/BambuStudio #3511: Gcode preview skips infill

brianharwell, 2024-01-30:
### Bambu Studio Version

1.8.4,51

### Where is the application from?

Bambu Lab github releases

### OS version

Window 11

### Printer

X1C

### How to reproduce

I slice a model, go to a layer, and preview the gcode steps.

[image]

There are only 606 instructions. Line 605 shows no infill.

[image]

Line 606 shows all of the infill.

### Actual results

Infill steps cannot be stepped through.

### Expected results

In previous versions (and in Orca) I can step through the infill steps.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-01-31:
Could your provide 3mf and gcode file?

brianharwell, 2024-01-31:
[Gridfinity Bins.zip](https://github.com/bambulab/BambuStudio/files/14114099/Gridfinity.Bins.zip)

Take a look at layer #77, I can't step through the infill. But I can step through the infill on layer #116.

LiZ-Li-BBL, 2024-02-01:
Hello,
The infill steps were reduced because of "Infill combination" parameter.
[image]

brianharwell, 2024-02-13:
@LiZ-Li-BBL @QingZhangBambu I reopened this ticket.

brianharwell, 2024-04-19:
@LiZ-Li-BBL @QingZhangBambu This is still not working. Please respond.

QingZhangBambu, 2024-04-22:
I will talk to the concerned staff. sorry to miss your message