bambulab/BambuStudio #3512: First Layer Not slicing

ECGUY5500, 2024-01-31:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Additional system information

i5 8th 12GB 2 TB SSD

### Printer

A1

### How to reproduce

10 flat parts on slice
change color of all parts with layer fill tool (100% color change)
Slice
Then the bottom layer of some parts don't slice with a bottom layer

### Actual results

[image]
[image]

### Expected results

They should be full parts with bottom on all, not some.  They slice fine before I change the color.  

### Project file & Debug log uploads

[Testing Bad Slice.zip](https://github.com/bambulab/BambuStudio/files/14112867/Testing.Bad.Slice.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2024-02-01:
for now, fix model would help you get right slice result

ECGUY5500, 2024-02-01:
What do you mean by fix model would help?

QingZhangBambu, 2024-02-02:
[image]

ECGUY5500, 2024-02-02:
I did the fix model and it changes the colors of the model to all one color.  Once I change the color back again it just reverts to the same model where the bottom layers are missing again.  I will add a new file to this chat.  
[Hockey testing.zip](https://github.com/bambulab/BambuStudio/files/14133542/Hockey.testing.zip)

QingZhangBambu, 2024-02-02:
I found that the elephant function makes some data miss mistakenly.