bambulab/BambuStudio #3513: Copy&Paste objects shuffles the order of object's modifiers and parts.

M-A-B007, 2024-01-31:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

WIN 10 Enterprise (ver 22H2)

### Additional system information

CPU:12th Gen Intel(R) Core(TM) i9-12950HX   2.30 GHz
RAM:64GB

### Printer

BambuLab X1C

### How to reproduce

1. Have an object with parts and/or modifiers.
2. Multiply that object on the plate by Copy & Paste.

### Actual results

[image]
The order of the copied part's sub-parts and modifiers is different than for the original object.

The slicer processes the modifier before it processes the main object now. It causes the modifiers to have no effect on the generated g-code.

### Expected results

Copy&Paste action does not change the order of the object's sub-parts and modifiers.

### Project file & Debug log uploads

[vetrani-kovovamrizka-zatisknuta.zip](https://github.com/bambulab/BambuStudio/files/14113236/vetrani-kovovamrizka-zatisknuta.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

MadMax389, 2024-01-31:
I can confirm this behavior. It plays havoc with the clipping order. Quite a pain to rearrange all the copies. PLEASE, PLEASE, PLEASE implement instancing like Prusa Slicer.

Haidiye00, 2024-02-02:
Hello, I have used your project and software version, but it has not been reproduced. Can you record a video to explain your issue? Thank you

MadMax389, 2024-02-07:
It is simple to reproduce. Add a cube primitive to the plate and then add a disc part (like a mouse ear)
[image]

When I slice it, the mouse ear has a higher clipping priority and will print inside the cube.
[image]

But I want the mouse ear to print outside the cube, so change the clipping order by rearranging the parts on the object screen.
[image]

Now copy or clone this assembly. The clipping order reverts back to the original clipping order.
[image]

This is really inconvenient when creating multiple copies of an object where I've changed the clipping order. You have to go to each model and rearrange them again. PLEASE, we are begging for instancing where changing something on one model propagates that change to all instances of that model (like Prusa Slicer). Thank you.

Haidiye00, 2024-02-18:
Thank you for reproducing this issue. We will find a way to solve it

Haidiye00, 2024-02-19:
[image]
I used our internal version 20240219 and there is no such issue. Please wait for version V1.9. Thank you

Haidiye00, 2024-05-14:
This issue has been fixed. Please download and use the latest version.

closed 2024-05-14 by Haidiye00