bambulab/BambuStudio #3517: Separation of 2 objects with different colors

jebsimmons, 2024-02-01:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS

### Additional system information

AppleM1
16GB RAM
MacOS Sonoma 14.2.1

### Printer

Bambu X1C or P1S

### How to reproduce

Insert 1 object. 
Insert 2nd Object- 1mm in height
Set the color for Object 1 to one color (Bambu matte Olive Green)
Set the color for Object 2 to a darker color (Bambu matte Black)
Center both objects
Assemble the two objects
Slice Plate

### Actual results

The software only recognizes object 1 and will not slice object 2.

### Expected results

It should slice the two objects and recognize that there are 2. This works on the Windows Version. 

### Project file & Debug log uploads

[SLICING 2 OBJECTS.3mf.zip](https://github.com/bambulab/BambuStudio/files/14124509/SLICING.2.OBJECTS.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

jebsimmons, 2024-02-01:
What it should look like: 

[image]

What it looks like after slicing:

[image]

zhimin-zeng-bambulab, 2024-02-02:
[image]

Because these two components have overlapping parts, We give priority to displaying the parts later in the object list.
If you want to display the black part, you can drag "part_2" behind "stl" in the object list, as show in FIG.

zhimin-zeng-bambulab, 2024-02-07:
@jebsimmons  Did the above method solve your problem?

zhimin-zeng-bambulab, 2024-02-07:
@jebsimmons We introduced this function in the third point of the release note of version V1.7.4 and gave a simple example.

closed 2024-02-07 by zhimin-zeng-bambulab