bambulab/BambuStudio #3521: Brim-object gap influences both object brim and support brim, no way to set separately -> mix of responsibilities.

byterendition, 2024-02-02:
### Bambu Studio Version

1.8.3.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab A1

### How to reproduce

1. A brim's responsibility is to hold on to an object, therefore brim-object gap is optimized to be as low as possible.
2. A support brim's responsibility is to hold the support to the base. The distance from this brim to the object is ALSO governed by the brim-object gap setting. Therefore, given point 1, the support gets stuck to the object on the first layer. This is a problem when trying to remove support from small holes.

Solution: link the support specific brim-object gap to the support/object xy distance setting or make it a separate setting altogether.

An example is given below:
[image]

### Actual results

Brim-object gap setting influences both object brim and support brim distance to the object.

### Expected results

Brim-object gap setting should be split in a setting for the object brim to object distance and a setting for the support brim to object distance.

Alternatively, the support brim to object distance should be linked to the support/object xy distance.

### Project file & Debug log uploads

[Bambu_Studio_brim_bug.zip](https://github.com/bambulab/BambuStudio/files/14138141/Bambu_Studio_brim_bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2024-02-05:
This is a specially designed feature: the bottom support acts like a brim to add on the adhesion, thus it should be controlled by the brim parameters. Of course we can remove this feature. but did you encounter any issues with it?

byterendition, 2024-02-05:
> This is a specially designed feature: the bottom support acts like a brim to add on the adhesion, thus it should be controlled by the brim parameters. Of course we can remove this feature. but did you encounter any issues with it?

Hi tangketan,

I wouldn't want to remove the feature, but split it up into one that acts on the object brim and one that acts on the support brim. I indeed had issues, where I needed the object brim to be a bit more firmly attached, while I didn't want this for the support brim, as it made it very hard to remove the support from small holes or shallow cavities on the bottom of my prints. I would suggest that the support tab gets its own brim settings. Also, the object brim and the support brim have a different color in the preview screen, indicating to me that these are already handled as different entities.

tangketan, 2024-02-06:
I see. I can split the option into two, but adding a different color for support brim would need more internal discussion, as there are alreasy too many colors.

byterendition, 2024-03-24:
> adding a different color for support brim would need more internal discussion, as there are alreasy too many colors.

I don't want to add more colors, I was stating that they are already colored differently, indicating that they are already handled differently in software. They just seem to be using the same brim gap option, and that is not correct in my opinion, as one has the function of holding on to my part while the other has the function of keeping the support brims away from the model.

So to summarize: I'd like to split the brim gap option into a brim gap option specifically for support-brims (could even be in the support tab) and a brim gap option for the 'normal' brim (located where it's now).

prentice78, 2024-04-15:
I'm also having this exact issue and request that this bug be fixed. It is *entirely* unclear that the BRIM-object gap should at all influence the SUPPORT first layer at all, because they are so different.

Related, another, lower-priority ask would be to have separate Brim-object gap settings for outer and inner brims.
But as the reporter mentions, the support gap is more critical to address.