bambulab/BambuStudio #3523: Addition auto generated support causes first layer slicer defect

ochiburi, 2024-02-02:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

MacOs Sonoma 14.2.1

### Printer

Bambu Labs P1S

### How to reproduce

Slice model with no support enabled, first layer looks perfect.
Enable auto generated support, slice model, first layer misses a section of the model.
[image]
[image]

### Actual results

Slice model with no support enabled, first layer looks perfect.
Enable auto generated support, slice model, first layer misses a section of the model.

### Expected results

Model should slice without defects

### Project file & Debug log uploads

Project file :
[Uploading Dog Bowl Riser 16cm v2.3mf.zip…]()

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-02-04:
[image]
the file could bot be download, could you upload a new one?

ochiburi, 2024-02-05:
[Dog Bowl Riser 16cm v2.3mf.zip](https://github.com/bambulab/BambuStudio/files/14163162/Dog.Bowl.Riser.16cm.v2.3mf.zip)