bambulab/BambuStudio #3527: Camera feed not working anymore

marcoroessner, 2024-02-02:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Win10

### Additional system information

[system.txt](https://github.com/bambulab/BambuStudio/files/14142306/system.txt)

### Printer

Bambu A1 AMS lite

### How to reproduce

open bambuslicer
open device
play camera feed

### Actual results

Black screen with -4 Error.
BUT it worked well in mobile app in PC app in and outside my network.
Now it works nothing. I did not change anythink.
All other data is shown on mobile app end pc app in and out my network

### Expected results

Camera feed shown.

### Project file & Debug log uploads

All projects.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-02-04:
@marcoroessner 
Maybe same as https://github.com/bambulab/BambuStudio/issues/3411#issuecomment-1911749637

seemly, 2024-02-06:
Quick question - could this be firewall related?

Could a firewall prevent a video feed from functioning?

bambu123, 2024-02-07:
The -4 error should not be firewall problem. Try reboot the printer.