bambulab/BambuStudio #3528: Underextrusion

marcoroessner, 2024-02-02:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Win10

### Additional system information

[system.txt](https://github.com/bambulab/BambuStudio/files/14142332/system.txt)

### Printer

A1 AMS lite

### How to reproduce

Last night I printed PETG file. print ok
today I replaced the modified models in bambu slicer and sent it and start printing.

### Actual results

The 1st layer was underextruded.
I cleand nozzle, then ok
[image]

### Expected results

Any error message from your flow sensor? For what is this when it does not measure underextrusion?

### Project file & Debug log uploads

[BB06-Motorhaus.zip](https://github.com/bambulab/BambuStudio/files/14142375/BB06-Motorhaus.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

amzaldua, 2024-02-12:
Are you printing with TPU? I see that you have modified the extruder lever. Normally, this modification is for printing TPU softer than recommended, so you may have your nozzle slightly stuck. Try unclogging it by following the Bambu wiki. You don't need to change the extruder lever again; the issue is not with that, it's a typical problem with TPU.

QingZhangBambu, 2024-03-06:
yes, please clean nozzle.