bambulab/BambuStudio #3529: Filament choice

Armanddeforce, 2024-02-02:
### Bambu Studio Version

1.8.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows

### Printer

x1c

### How to reproduce

1 Change filament to #3 #4 for multi filament slot 1 and 2 filled with asa set these for PLA
2 slice object
3 Print object the screen displays only slot #1 as pla but has the appropriate color for for slot 3 other filaments in AMS are greyed out and unable to be selected 

The only way to correct this is to switch the filament to slot 1 this does the same with colors also which makes the AMS almost useless with multi material 

### Actual results

It will print with slot one with the incorrect settings (PLA settings for ASA) and is the incorrect one despite attempts to change it.

### Expected results

when setting the filaments to print I should be able to chose any slot to print from and not have it changed post slice in the printing or having to change the filament to slot #1 in the AMS and it should display the correct color for the slot (which it does but displays the incorrect number for the slot it has chosen for me.)

### Project file & Debug log uploads

[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-02-04:
Could you upload the 3mf file and gcode file?

Armanddeforce, 2024-02-05:
This happens with any file used to print not just a specific one.  To test this Put ASA in the first slot but try to only print from the other 2-4 with PLA now on the new version it gives an error that the temperature difference is to grate to print with. Meaning that all the filaments have to be the same type and you can not start a print wit out using slot 1 first.  This means you have either to repaint your object in the order of slots or reload the filament to slot 1 with base filament. Even with a basic sign made on Maker world tis will occur.Paul

walterwongbbl, 2024-02-09:
@bam-winterfell

walterwongbbl, 2024-05-21:
Please use our latest 1.9.1 version

closed 2024-05-21 by walterwongbbl