bambulab/BambuStudio #3530: Line Width Color Scale Bounds

CCS86, 2024-02-03:
### Bambu Studio Version

1.8.3

### Where is the application from?

Bambu Lab github releases

### OS version

Win 10 x64

### Printer

X1C

### How to reproduce

Slice a print, choose color scheme: line width

### Actual results

Line width color scale bounds are far outside any visible lines in the print. This renders half the color scale useless.

### Expected results

I expect that the maximum on the line width scale matches the widest line in the print. 

Ideally, the automatic scale would be bounded by the widest and narrowest lines in the print, and also allow the scale bounds to be manually changed.

### Project file & Debug log uploads

.....

### Checklist of files to include

- [ ] Log file
- [ ] Project file

CCS86, 2024-03-12:
Improvements to the legend added in the latest Super Slicer:

https://github.com/supermerill/SuperSlicer/releases/tag/2.5.59.8

LiZ-Li-BBL, 2024-03-25:
Thanks for your suggestion!
We will consider this improvements in the future release!

CCS86, 2024-05-04:
@LiZ-Li-BBL any updates on this issue?

CCS86, 2024-06-25:
@LiZ-Li-BBL?

CCS86, 2024-07-06:
@QingZhangBambu Is this issue going to be resolved?

QingZhangBambu, 2024-07-08:
@CCS86 I talked to the concerned staff. It's already in our feature pool, however, not enough staff to implement it.

QingZhangBambu, 2024-07-08:
It would be great if you or others would like to pull a request on github. I will change this tag from bug to feature.

CCS86, 2024-07-08:
I really wouldn't consider this a "feature". It is a bug in the existing color scale feature.

What is the justification of having the color scale set the maximum automatically to a width 3 times larger than any width in the sliced print?