bambulab/BambuStudio #3531: kleur werkt niet

rickert1, 2024-02-03:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Built from source by yourself or others

### OS version

windows 10

### Printer

bambu lab x1-carbon

### How to reproduce

als ik de kleur gevuld heb in het bestand die ik wil en als ik dan ga slicen dan doet hij de andere kleuren er niet in.

### Actual results

[image]
[image]

### Expected results

dat dit probleem wordt opgelost

### Project file & Debug log uploads

lukt niet

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-02-04:
@rickert1  The link you gave cannot download the file, could you share your 3mf file?

zhimin-zeng-bambulab, 2024-02-07:
@rickert1  hello, could you share your 3mf file?

zhimin-zeng-bambulab, 2024-02-21:
@rickert1  Could you share your 3mf file?

closed 2024-08-20 by zhimin-zeng-bambulab