bambulab/BambuStudio #3533: BambuStudio connect multiple accounts to one printer

hutje, 2024-02-03:
Being able to add other users to access the printers from within BambuStudio.
In an educational environment such as a school or library it is needed to give multiple users access to the printers. With the ability to give other users some privileges to send their 3D-print to the printer or seeing the live footage.

The owner can give different users different privileges, for example: controlling the printer, sending sliced objects (with or without remote starting a print privilege) or viewing the livestream.

Giladkapel, 2024-07-08:
I would also like to see this feature implemented.

Jason-Mefford, 2024-09-13:
Yes bambu this would be very nice to have!

Habasch, 2024-10-15:
Please finally implement this function. Creality has done it too. And there are so many people waiting for it.

luckygreen, 2024-10-26:
Multiuser functionality on the X1C is **way overdue**. When will this be implemented?

LarryHiller, 2024-10-29:
I bought two x-1 carbons for my classroom.  I dont want to have to make a single account that everyone shares.  Please allow everyone to sign into a machine with their own account.

Excds, 2024-11-04:
It would be very practical to have this implemented.

MartinPavlicek, 2024-11-15:
This would be a very useful feature in many scenarios.

TheSlider2, 2024-11-20:
+1
We have multiple printers at work and I have mine at home. It's a pain to switch accounts each time I need to access one. 
This feature probably also requires the firmware to list multiple authorized accounts in order to also be able to "forget" them.

napalmz, 2024-12-20:
+1
Also, multi account login on Bambu Studio (or retain login credentials to fast switch between accounts managing printers).
My use case is: Office Print with corporate account and personal printer with personal account.

dahnutz, 2025-01-23:
I subscribe, definitively need multi user support, or share with other account/s or at least dual support for cloud and lan access.

NoahK216, 2025-02-08:
+1
Please make this happen!

AeroDynamicDuck, 2025-04-02:
+1 to this! Entirely agree. It's silly to have to unbind the printer between accounts whenever switching.
Or share an account, when the other person doesn't want what I'm printing to affect their feed/history.

apiominer, 2025-04-02:
Agreed, I'd like to let the rest of my fam use the A1 mini to cloud print.  I'd definitely go through more filament that way--just saying! 😉

a-well, 2025-04-10:
+1 Please, pretty please

GuyH77, 2025-04-11:
Please add this

PIZZA-ALERT, 2025-05-22:
ADD IT ALREADY BAMBU FFS

Giladkapel, 2025-07-13:
Creality had the same issue and they added additional 2 users. One is master and there can be 2 more "secondary" users. While this isn't a full solution, I would say it answers for 90% of the multi-user cases.

TraXie666, 2025-11-13:
We really need this feature!

Rupp3r, 2025-11-14:
@DanBao-Bambu any chance that you have some info on the status this feature or could have someone look at this?

Pretty old case and a VERY popular demand. Think the only reason it have not had more interaction before now, is because we all expected it to happen soon and did not want to spam. Same with likes - it is such a common feature that I think most people expected it to be done whether liking the suggestion or not(and it is still the 6'th most liked case and 5'th liked feature suggestion..)

DanBao-Bambu, 2025-11-15:
Using the LAN mode access of the printer should be able to solve the problem of multiple people sharing the printer, provided that it is under a unified LAN.
@walterwongbbl Please check this requirement.

luckygreen, 2025-11-17:
@walterwongbbl wrote:
> Using the LAN mode access of the printer should be able to solve the problem of multiple people sharing the printer, provided that it is under a unified LAN.

Neither I nor anyone else that wants this feature, who have bought into the Bambu ecosystem, has any interest in having `to` use LAN mode to share a printer within a family or students in a classroom. (Note that there are obvious follow-up requirements, in particular for classroom use and use within a family. You need at least two roles: regular user (student/child) and an approver role (teacher or parent) with the ability to approve sending an object to print and to perform queue management. I.e. "print _that_ object next, even though it is 12th in queue".

bencardinal, 2025-12-29:
100% what @apiominer said!  I recently added a second Bambu printer and would love to be able to share with my child and help teach how to use it without completely changing the account or sharing the account.

kusmi, 2025-12-29:
Yep - I currently share the account with my child and while it works ok, I now have 100 likes and collected item in my account from my child and can’t find my saved models from makerworld 😉