bambulab/BambuStudio #3534: heat bed not heating up

Hradecky123, 2024-02-03:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows11

### Additional system information

heat bed not heating up. On the display is 24 C

### Printer

Bambulab X1 Carbon

### How to reproduce

During my second printing with PETG is temperature of the plate falled slowly down.

### Actual results

heat bed not heating up

### Expected results

?

### Project file & Debug log uploads

?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-02-04:
could you provide the 3mf and gcode file for us to target the issue?

Hradecky123, 2024-02-04:
For the printing, when the plate stoped heating  i used attached step file. 

> Od: "Shikieiki" ***@***.***>
> Komu: "bambulab/BambuStudio" ***@***.***>
> Datum: 04.02.2024 07:39
>
> CC: ***@***.***>
 
could you provide the 3mf and gcode file for us to target the issue?

fly2016-1-12, 2024-02-04:
hi,

fly2016-1-12, 2024-02-04:
Hi, can you help raise a tikect on the handy APP and give us the tikect number, we need to combine the printer logs together to see the issue. Thanks~

Hradecky123, 2024-02-04:
Log from mobile app uploaded to US240203753001

Hradecky123, 2024-02-06:
[image]
[image]
Hi, I have more detailed information. My first description was a bit vague
AC board voltage output is OK. See attached picture
The heating bed resistance value does not appear on multimeter!!! See attached picture.

Hradecky123, 2024-02-10:
Hi, can you tell me please, how long must i wait for help.