bambulab/BambuStudio #3535: Fuzzy skin painting tool

megafonmann, 2024-02-03:
**Is your feature request related to a problem? Please describe.**
Working with the fuzzy skin feature can be hard, especially for beginners. Unwanted fuzzy skin areas need a bit of workarounds to be achieved.

**Describe the solution you'd like**
A new function in the color painting tab [N] to fill areas that should the fuzzy skin texture on it.

**Describe alternatives you've considered**
Current workarounds would be adding modifiers which apply the fuzzy skin onto those areas.

**Additional context**
An example would be the creation of objects with a logo/inscription, but these are lost due to the "fuzzy skin" function. With the new function you could either select and block the areas of the logo, or select the walls to which the "fuzzy skin" function should be applied.

vincentbcd, 2024-03-28:
Yes, very good idea!

Hardrada, 2024-09-06:
For the love of all that is holy.. please add this feature. It would be an overnight game changer.

magnusmyklebost, 2024-11-27:
Please implement the stuff this genius has done:
https://www.youtube.com/watch?v=cNkHfydnUCI

magnusmyklebost, 2024-11-27:
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1

closed 2026-02-04 by megafonmann