bambulab/BambuStudio #3536: A1/A1 mini - prints in "calibration flow area" BUG

Eldenroot, 2024-02-04:
### Bambu Studio Version

1.8.4

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Win11 64bit

### Additional system information

---

### Printer

A1

### How to reproduce

Auto arrange objects - they are placed in the "calibration" area... which is not good and it could cause a lot of troubles.

### Actual results

This option for A1 family is missing in "auto arrange object"

[image]

[image]

[image]

[image]

### Expected results

It should be here, because now it prints in "calibration" area. Which is not good. Check the image!

### Project file & Debug log uploads

---

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-02-05:
we will improve that.

walterwongbbl, 2024-04-10:
Please use the latest firmware version for verification

closed 2024-10-17 by walterwongbbl