bambulab/BambuStudio #3537: TPU 95A AMS lite update

marcoroessner, 2024-02-04:
If I use TPU 95A in my AMS lite and the orange extruder wheel fix (pushing the counterwheel more to the extruder gear wheel) and I push 1cm more the TPU filament in the bowden, the change works.

For TPU use please upgrade the AMS that it pushes more in than out. The flex needs to be compensated.

After changing to TPU the AMS lite pushes the fila to the extruder, the extruder rotates and checks , no fila loaded, the AMS pulls TPU out.
At the 2nd try I push 1cm more the fila into the AMS inlet and the change works.

marcoroessner, 2024-02-04:
And please do not tell my that AMS lite is not compatible with TPU, this is not true. Where is your wish to get clever problems solved?! I tought you where Ingenieurs too. 
So please solve the problem by adding a TPU profile to our AMS. And change the preload of the AMS lite Extruder counterwheel so that TPU is not squished to hard...

80sFPV, 2024-09-11:
I would recommend some kind of conversion kit with a different set of springs (softer) and slightly larger feeding gears (larger diameter to compensate for the compression of the TPU).