bambulab/BambuStudio #3538: [Linux] Some text unreadable in certain situations (Ex. Tabs if not focused). Some unreadable at all times.

MostHated, 2024-02-04:
### Bambu Studio Version

1.8.4.51 and earlier

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 23.10

### Additional system information

<details>
<summary>System Details</summary>

OS: Ubuntu 23.10 x86_64
Host: X670E Taichi
Kernel: 6.5.0-15-generic
Uptime: 3 days, 1 hour, 57 mins
Packages: 4502 (dpkg), 471 (nix-user), 48 (nix-default), 82 (flatpak), 26 (snap)
Shell: zsh 5.9
Resolution: 2560x1080
DE: GNOME 45.2
WM: Mutter
WM Theme: Adwaita
Theme: Yaru-blue-dark [GTK2/3]
Icons: Yaru-blue [GTK2/3]
Terminal: alacritty
CPU: AMD Ryzen 9 7950X (32) @ 5.881GHz
GPU: AMD ATI Radeon RX 6800/6800 XT / 6900 XT
GPU: AMD ATI 5a:00.0 Raphael
Memory: 11982MiB / 63398MiB

</details>

### Printer

Bambu Lab x1c

### How to reproduce

Open various menus that contain tabs, for example, the filament options menus, seen below.
Initially, the first selected tab has unreadable text, but if you select the second tab, the first tab is readable. If you select the first tab again, it becomes unreadable once again.

The second is when opening additional static windows with no tabs. Much of the text is simply unreadable at all times. Such as the printer options menu:

### Actual results

Unreadable First Tab Text
[image]

Readable First Tab Text
[image]

Unreadable Printer Options Menu
[image],

### Expected results

All text on all interfaces and menus should be readable.

### Project file & Debug log uploads

No project file is needed. This happens even when opening a blank/new BambuStudio instance. I am currently using the flatpak, which doesn't seem to create a log file at the location specified ($HOME/.config/BambuStudio/log). I only have logs there from the non-flatpak version. Though, that version also has the same issue.

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2024-02-09:
@MostHated 
It seems that our dark mode is not working properly on Linux.

jmarmstrong1207, 2024-04-13:
Any updates on this?

MostHated, 2024-04-13:
I am still having the issue in many places. For example the Objects menu:

[image]

jmarmstrong1207, 2024-04-13:
Yeah I'm having the same issues. Main things bugging the Linux build is this and all the dependencies not packaged in the AppImage

walterwongbbl, 2024-05-06:
@jmarmstrong1207 @MostHated 
The color mapping in dark mode is incorrect, and we have made some optimizations. An update will be available soon.

closed 2024-10-17 by walterwongbbl

go-hacks, 2025-08-16:
This is still an issue in the latest version and has not been fixed. I've even switched to different themes in my DE and it makes no difference. Even with white windows and black fonts they do not show up. I can't even configure my printer because of it.