bambulab/BambuStudio #3540: Timelapsvideos are splitted into .bin files

BienesParaTex, 2024-02-05:
### Bambu Studio Version

1.8.5.41

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambulab A1 Combo

### How to reproduce

1. I printed a part with timelaps and as it is not possible to get on the sd card to watch them I copied them to the PC.
2. Those files are no AVI or MP4 files but .bin files
3.  I tried to play them with VLC and some other players but they are not working.
4. I  tried to convert them - and no luck. They are still not working.

### Actual results

I gave up and searched for hours in the World Wide Web but still no answer to the question "How can I convert them to AVI or better MP4.

### Expected results

I would love to have a version of the files that I can  watch on a player or upload for example to TicToc.

### Project file & Debug log uploads

[image]
[CMN_03919A3B2100366_02-04_15_50_20.779_idx_92.zip](https://github.com/bambulab/BambuStudio/files/14166971/CMN_03919A3B2100366_02-04_15_50_20.779_idx_92.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

bambu123, 2024-02-06:
It should be that timelapse was not generated and has nothing to do with the bin file. This may be a printer problem. Can you help submit a ticket and attach the printer log?

bambu123, 2024-02-06:
Also, please confirm if timelapse is enabled.

BienesParaTex, 2024-02-06:
Hello, yes the timelaps Standard was enabled. Unfortunately I can't check
anything else actually because of the recall problem.

But timelaps worked at a previous print  But with the smooth option.

Kind regards

Sabine

BienesParaTex, 2024-02-06:
Hello together,

I am so sorry. It was my mistake 🙈

I copied the files from the wrong folder recorder instead of timelaps.

So you can close the ticket. And sorry again.

Kind regards

Sabine Wagner