bambulab/BambuStudio #3541: Lost videofeed after network drops

mctazman, 2024-02-05:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 14.3 (23D56)

### Printer

X1 carbon, firmware 01.07.00.00

### How to reproduce

I had some network problems, which made me power cycle the AP where my X1 is connected to. After the network comes back online, I can not get any video feed. No matter the device, software, or OS. After a power cycle, the video feed works perfectly.

It seems to me, that the video feed has some network binding that doesnt get reset when the machine loses its connection. Probably a token in cache, if I should hazard a guess

This is the second time I experienced this problem. Last time was on an older firmware version.

### Actual results

The software keeps loading, until it times out, starts loading the feed again, repeat.

### Expected results

Preferably, check for network drops & re-initialize the video feed if possible.

### Project file & Debug log uploads

[Archief.zip](https://github.com/bambulab/BambuStudio/files/14168594/Archief.zip)
[Speaker mount.3mf.zip](https://github.com/bambulab/BambuStudio/files/14168596/Speaker.mount.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2024-02-06:
We recently released a web plugin 01.08.04.04 related to this issue. Have you upgraded?

bambu123, 2024-02-06:
> We recently released a network plugin 01.08.04.04 related to this issue. Have you upgraded?

mctazman, 2024-02-06:
I think I've upgraded. I got the message & clicked "ok", is there a way to check? I upgraded before the issue started, so restarting Studio did not fix it.

bambu123, 2024-02-06:
The printer may need some time to restore its network connection.
Could you help provide the Studio logs?
Also, please tell me the approximate time when the problem occurred.

bambu123, 2024-02-06:
> The printer had hours, and no feed returned. The logs are in the Archief.zip file, in the original post. I don't know the time, it was morning. If I were to guess, I'd say around 6-ish. But truly have no idea.

I think it should be a problem with the printer. The printer has an occasional crash problem which may be fixed in the coming version 1.8.
However, To figure out the problem
Could you help to create a ticket and upload the log (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log).
Please tell me the ticket number, after you create the ticket.

mctazman, 2024-02-06:
Thanks for your quick reply. The ticker number is: US240206361001

bambu123, 2024-02-06:
> Thanks for your quick reply. The ticker number is: US240206361001

OK. Our printer team will check it.

bambu123, 2024-02-06:
@mctazman 
From our printer team:
Same As https://github.com/bambulab/BambuStudio/issues/3199#issuecomment-1909439238.

mctazman, 2024-02-06:
Cheers, very clear. Thanks for the quick reply & work.

closed 2024-02-06 by mctazman