bambulab/BambuStudio #3542: Missing dark mode option with flatpak and unable to enable with config file

g4njawizard, 2024-02-05:
### Bambu Studio Version

 v01.08.04.51 

### Where is the application from?

Bambu Lab github releases

### OS version

Debian

### Additional system information

Using XFCE. To start BS i have to use a terminal

### Printer

X1C

### How to reproduce

1. Install and launch Bambu Studio using Flatpak
2. Open preferences or open configuration folder
3. Try enabling dark mode

### Actual results

The Preference Window does not show this option "dark mode"

changing the config line from "0" to         "dark_color_mode": "1", has no effect. Relaunching the application overwrites the config.

[image]

### Expected results

Dark Mode enabled

### Project file & Debug log uploads

[debug_Mon_Feb_05_18_01_56_4.log](https://github.com/bambulab/BambuStudio/files/14170037/debug_Mon_Feb_05_18_01_56_4.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bam-winterfell, 2024-02-06:
@g4njawizard On many Linux systems, BambuStudio follows the system color. Would you like to try switching system colors ？

g4njawizard, 2024-02-06:
Doesn't work for me. Systemsettings are already on dark mode. Same version as Appimage has no trouble. It's just the flatpak one.

rrregit, 2024-02-07:
@g4njawizard Okay, we will try to reproduce this issue

bam-winterfell, 2024-02-07:
@g4njawizard Okay, we will try to reproduce this issue

BradleyUffner, 2024-02-21:
Exact same issue for me.

Ubuntu 23.10
KDE Plasma 5.27.8 (in Breeze Dark theme)
Bambu Studio 1.8.4.51

LukasVrox, 2024-03-13:
Same here.

Nobara Linux 39
KDE Plasma in Breeze Dark Theme
Bambu Studio 1.8.4

BradleyUffner, 2024-03-19:
Still not fixed, so I decided to switch to Orca Slicer, which is based on Bambu Studio with additional features, including a working dark mode.

aslwerner, 2024-03-19:
Try this way it works for me fine with Ubuntu 23.10
**(https://github.com/bambulab/BambuStudio/issues/2567#issuecomment-2007302233)**

It should help 
[image]

BradleyUffner, 2024-03-19:
> Try this way it works for me fine with Ubuntu 23.10 **([#2567 (comment)](https://github.com/bambulab/BambuStudio/issues/2567#issuecomment-2007302233))**

Thanks, that worked, but I think I'm still going to stick with Orca Slicer for now. I'm enjoying the extra features like the scarf seams.

walterwongbbl, 2024-03-22:
@BradleyUffner  
Please follow our upcoming version, thank you.

walterwongbbl, 2024-03-29:
@BradleyUffner 
We currently do not have manpower to solve the support for dark mode, but you can already use scarf seams.

flying-x, 2024-05-05:
It is said I had to use flatpak on manjaro/arch. It worked pretty well a few month back. Now, only bambustudio-git works but video streaming doesn't work even with that. So I had to use flatpak for streaming. Then, the theme option wasn't there... Orca slicer wasn't better for me on arch either. It stuck at the first screen of selecting user's region.

hadess, 2024-10-09:
The Bambu Studio Flatpak update that's coming later today will follow the system theme configuration.

The fix is here:
https://github.com/flathub/com.bambulab.BambuStudio/pull/16

closed 2024-10-17 by walterwongbbl

Arnowt, 2025-02-18:
Unreadable color, and I can't turn off the dark theme.

[image]

go-hacks, 2025-08-16:
Yes, this issue is NOT fixed. Changing the system theme does not make the text readable no matter what color it is. Changing the config does not fix it and there is no dark mode option in the preferences to enable or disable. I am using the latest flatpak from flathub on XFCE with various GTK themes to no avail. Orca Slicer does not have this problem. Dark theme or no dark theme so whatever they are doing is the solution. This is not the first application I've seen that can't figure out GTK text colors yet many do not have any problems with it no matter what theme you have so there IS a solution. Someone well versed in GTK would probably laugh at all of this :)

ThatOhio, 2025-08-17:
Have not been able to get dark mode to work in the flatpak for some time now. Tried the metadata edit linked in another comment and everything else I could find and no luck.

ddmgh, 2025-11-14:
Just wanted to say a recent Plasma update has broken this for me as well; the BS flatpak is not longer inheriting the system theme.

PC-Heini, 2025-12-03:
The only fix that works for me is was adding `GTK_THEME=Adwaita:dark` to the environment variables of the flatpak.
I guess every other dark GTK theme would work as well.

evil666, 2025-12-19:
> The only fix that works for me is was adding `GTK_THEME=Adwaita:dark` to the environment variables of the flatpak. I guess every other dark GTK theme would work as well.

The fact that this is still not fixed boggles my mind. However this worked for me by only adding this to the metadata file. Running latest fedora and the application is flatpak. Thanks for this!

go-hacks, 2026-01-11:
I'll have to try that. I can't remember if I tried it that way but it MIGHT work by sending config to the flatpak launch. It's still a bug though and it's not implemented properly. The flatpak should be providing its own properly configured dependencies and if not, it needs to allow people to use their local ones. At the very least it should be able to detect the local requirements and choose whatever launch options are necessary to display properly on its own.

Changing the host theme doesn't fix the problem. It changes the theme from dark/light but the text is unreadable in either. Almost like it's half providing its own GTK stuff and half relying on the host but should be 100% one or the other.

In comparison, Orca works just fine using their appimage and my local configuration so the proper way to handle all this is definitely in their codebase and should be implemented the same way here. I've seen EXACTLY this problem in many apps over the years and it's ALWAYS because it wasn't built right. Not user error. It can happen even with locally installed apps and not appimage/flatpak.

Sent from [Proton Mail](https://proton.me/mail/home) for Android.