bambulab/BambuStudio #3543: Add display of set nozzle size from firmware in Bambu Studio

sprior, 2024-02-05:
**Is your feature request related to a problem? Please describe.**
If you change nozzles it's easy to forget what you've got in there when you're sitting at Ba,bu Studio and not next to the printer.

**Describe the solution you'd like**
The machine LCD panel in the maintenance page lets you set the nozzle size and whether it's hardened or stainless.  It would be nice if Bambu Studio could query that info and display it in the machine status screen.

walterwongbbl, 2024-02-06:
@sprior 
coming soon

sprior, 2024-02-06:
Awesome!

sprior, 2024-02-06:
If Bambu Studio checked that value so it pops up a warning if the nozzle size in BS doesn't match what's set in firmware that would also be great.