bambulab/BambuStudio #3544: CLI - Arrange specific STL files at specific positions of multiple plates from the command line - Assemble list JSON

abdullahtahiriyo, 2024-02-05:
I am in need to programmatically generate from the command line some 10 plates of objects, each plate having three different objects which need to be arranged in a very specific position (auto arrange won't work in this case, I cannot share the model).

A possible way to do it with what is in place might be via:
./BambuStudio_linux_ubuntu_v01.08.04.51_20240117.AppImage --load-assemble-list assemble_list.json

This option is however not documented. I have only seen references to it in the source code. It looks it has the potential:

```
#define JSON_ASSEMPLE_PLATES                   "plates"
#define JSON_ASSEMPLE_PLATE_PARAMS             "plate_params"
#define JSON_ASSEMPLE_PLATE_NAME               "plate_name"
#define JSON_ASSEMPLE_PLATE_NEED_ARRANGE       "need_arrange"
#define JSON_ASSEMPLE_OBJECTS                  "objects"
#define JSON_ASSEMPLE_OBJECT_PATH              "path"
#define JSON_ASSEMPLE_OBJECT_COUNT             "count"
#define JSON_ASSEMPLE_OBJECT_FILAMENTS         "filaments"
#define JSON_ASSEMPLE_OBJECT_POS_X             "pos_x"
#define JSON_ASSEMPLE_OBJECT_POS_Y             "pos_y"
#define JSON_ASSEMPLE_OBJECT_POS_Z             "pos_z"
#define JSON_ASSEMPLE_OBJECT_ASSEMBLE_INDEX    "assemble_index"
#define JSON_ASSEMPLE_OBJECT_PRINT_PARAMS      "print_params"

typedef struct _assemble_object_info {
    std::string         path;
    int                 count;

    std::vector<int>    filaments;
    std::vector<int>    assemble_index;
    std::vector<float>  pos_x;
    std::vector<float>  pos_y;
    std::vector<float>  pos_z;
    std::map<std::string, std::string> print_params;
}assemble_object_info_t;

typedef struct _assemble_plate_info {
    std::string         plate_name;
    bool                need_arrange {false};
    int                 filaments_count {0};

    std::map<std::string, std::string> plate_params;
    std::vector<assemble_object_info_t> assemble_obj_list;
    std::vector<ModelObject *> loaded_obj_list;
}assemble_plate_info_t;
```
But I would need a proper assemble_list.json file, as otherwise, reverse engineering it would take me some time.

So, I would be grateful if you could give me some orientation about how to do what I want to do, and whether this option could be helpful (and if so an example JSON code would be very welcome). If you see an alternative way of doing what I want it is also very welcome.

Thanks!!

abdullahtahiriyo, 2024-02-15:
@lanewei120

Any chance to get a sample assemble-list.json?

lanewei120, 2024-02-16:
@abdullahtahiriyo 
yes, we will share you a sample json after we returned to work(currently on our holidays)

abdullahtahiriyo, 2024-02-16:
新年快乐

lanewei120, 2024-02-18:
@abdullahtahiriyo 
thanks

please find it from
[assemble_list_sample.json](https://github.com/bambulab/BambuStudio/files/14320982/assemble_list_sample.json)

dd-dkenney, 2024-02-22:
@lanewei120 hello again - I just wanted to confirm if the `path` defined in the assembly list JSON you provided is able to be set as an absolute path to the file on the filesystem somewhere?

lanewei120, 2024-05-14:
oops, just missed this thread before
the path can be an absolute path and I think you should already verified it

praetp, 2024-05-21:
Can you share a fully working example ?

lanewei120, 2024-05-22:
> @abdullahtahiriyo thanks
> 
> please find it from [assemble_list_sample.json](https://github.com/bambulab/BambuStudio/files/14320982/assemble_list_sample.json)

here is the example input file, just run following command:
./bin/bambu-studio --export-3mf out.3mf --load-settings="machine.json;process.json" --load-filaments="pla_basic.json;pla_basic.json;pla_basic.json;pla_basic.json" --filament-colour="#FFFFFF;#0000FF;#00FF00;#FF0000" --load-assemble-list="assemble_list_sample.json"

rizwansarwar, 2024-08-23:
@lanewei120 is this still working? I just opened a bug because this process fails for me. I am using pretty much same command as above. see #4666

lanewei120, 2024-08-24:
it should still work
do you mean it worked before, and can not work now?
@rizwansarwar

rizwansarwar, 2024-08-24:
@lanewei120 I personally never had it working. I was asking because you posted a couple of times implying it was and is working for you.

lanewei120, 2024-08-24:
you need to use the full config file
[image]

rizwansarwar, 2024-12-31:
@lanewei120 @abdullahtahiriyo Did you guys try to use height ranges in assemble_list_sample.json? I can't seem to get them to work. I have tried to specify it as follows and it seems to match with the structure required for json. The slicer slices and even detects that height ranges have been specified, however, when I open the generated .3mf in slicer gui. The height ranges are missing. Wondering if it is a bug? Would really appreciate help to figure this out.

[image]

[assemble-list-sample.json](https://github.com/user-attachments/files/18282808/assemble-list-sample.json)

lanewei120, 2025-01-01:
> @lanewei120 @abdullahtahiriyo Did you guys try to use height ranges in assemble_list_sample.json? I can't seem to get them to work. I have tried to specify it as follows and it seems to match with the structure required for json. The slicer slices and even detects that height ranges have been specified, however, when I open the generated .3mf in slicer gui. The height ranges are missing. Wondering if it is a bug? Would really appreciate help to figure this out.
> 
> ![slice_data](https://private-user-images.githubusercontent.com/3135402/399505333-c0154bd7-890b-4b73-9f02-935a50109b29.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU3MDQwODYsIm5iZiI6MTczNTcwMzc4NiwicGF0aCI6Ii8zMTM1NDAyLzM5OTUwNTMzMy1jMDE1NGJkNy04OTBiLTRiNzMtOWYwMi05MzVhNTAxMDliMjkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEwMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMDFUMDM1NjI2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzVmOGViN2M2ZGY0Nzg1NWY4NDIyMTE3MzBlM2YyN2FhMDRjMzU1MTUwYjA1ZDcyNDMwMDA2MmVhMzg2NmRhMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.8CR9w3IDnlwH6enSvImWkFFs_FdnbEaHypFuHWvrm00)
> 
> ![missing_height_ranges_in_slicer](https://private-user-images.githubusercontent.com/3135402/399505753-d0deef51-2cc8-43d8-9452-81e3a0b72f63.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU3MDQwODYsIm5iZiI6MTczNTcwMzc4NiwicGF0aCI6Ii8zMTM1NDAyLzM5OTUwNTc1My1kMGRlZWY1MS0yY2M4LTQzZDgtOTQ1Mi04MWUzYTBiNzJmNjMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEwMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMDFUMDM1NjI2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9M2ZlN2Y0ZTliYjU5ZDZmMDdjMmY3YjQ4OGIyYjc3OGQ2YmUxOGY3NjNhZTkwNmI5ZjIxZGIwOTBlYjRjOGNjNiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.n_d3D4uykfaS0xkJwwMdp5n2K7lN_c1RtngDnCUwYy4)
> 
> [assemble-list-sample.json](https://github.com/user-attachments/files/18282808/assemble-list-sample.json)

for height_range
[image]
we need to always add "extruder" and "layer_height" in it

rizwansarwar, 2025-01-01:
@lanewei120 Thank you for quick response. I tried your suggestion. I still did not work. I am thinking maybe the location where I am specifying range_params is not valid? I am using the latest stable release of BS on Ubuntu 24.04. Could you kindly look at my Assembly json attached and see if there is something wrong in it? Many thanks

[assemble-list-sample.json](https://github.com/user-attachments/files/18285112/assemble-list-sample.json)

lanewei120, 2025-01-01:
just check the codes, and find that height_range can only be used in assembled_params now
it is only used in an assembled object
please refer to following example:
[image]

rizwansarwar, 2025-01-01:
@lanewei120 That worked exactly as I expected it to. Thank you so much. One thing I still can't figure out is how to specify color change at a layer height. In the slicer, I can do on preview stage. However I am not sure how to do that via CLI. But more importantly for assembled objects, how do I map different filaments that are loaded from command line?

lanewei120, 2025-01-03:
@rizwansarwar 

it seems we don't support the color changes in gcodeview under the cli mode

you can identify the colors of each sub-objects in the assembled object using "extruder"

tqtb89, 2025-09-24:
> [@rizwansarwar](https://github.com/rizwansarwar)
> 
> it seems we don't support the color changes in gcodeview under the cli mode
> 
> you can identify the colors of each sub-objects in the assembled object using "extruder"

How to slice multi-color object with AMS via cli?

I just successfully generated gcode for 2 filament, but it's not printable.

`M620 S[initial_no_support_extruder]A   ; switch material if AMS exist`
initial_no_support_extruder is always 0, cause AMS choose wrong filament and got stuck when switching material

Here's example:
```
    ./Bambu_Studio_ubuntu-22.04_PR-8184.AppImage     --orient 0     --arrange 1     --scale 3     --load-settings "/tmp/tmpv05or22v_machine.json;/tmp/tmpskd93pz__process.json"     --load-filaments "/tmp/tmpt5kyia5h.json;/tmp/tmp14fb6k9n.json"     --load-assemble-list "/tmp/tmpnsx3wi1b.json"     --filament-colour "#FFFFFF;#FFFFFF"     --slice 0     --debug 4     --export-3mf /tmp/tmph87thjv_     
```

[tmpskd93pz__process.json](https://github.com/user-attachments/files/22516444/tmpskd93pz__process.json)

[tmpv05or22v_machine.json](https://github.com/user-attachments/files/22516466/tmpv05or22v_machine.json)

[tmpnsx3wi1b.json](https://github.com/user-attachments/files/22516486/tmpnsx3wi1b.json)

[tmp14fb6k9n.json](https://github.com/user-attachments/files/22516493/tmp14fb6k9n.json)

[tmpt5kyia5h.json](https://github.com/user-attachments/files/22516497/tmpt5kyia5h.json)