bambulab/BambuStudio #3546: BambuLab Studio - crash

Maci68, 2024-02-06:
**Is your feature request related to a problem? Please describe.**
MasOS Sonoma, after sending the job via the cloud, the application crashes, it is possible to reopen

netechsys, 2024-02-07:
Same with me as well..  After this update, I am constantly getting crashes.  Another thing that started is I get strange graphic artifacts on the screen.  I have reinstalled the OS from scratch.. and only installed Office and Bambu Studio, and the strange issue is still there..  I think I will reinstall again with nothing, but this is a new (45 days old) M3 Max MacBook Pro..  I have checked the hardware using the Diagnostic test, and all tests good, so it is a software issue...  Strange that it started after I think the last Network Plugin update.

robbinonline, 2024-03-03:
Same issue here.

mrunion, 2024-03-11:
FWIW: I get occasional crashes with Bambu Studio (version 1.8.4.51) on my M1 iMac. It seems to always happen after sending a job to the printer and Bambu Studio is closing the print dialog and switching over to the Device View.

The Mac crash report is below. To my untrained eye it seems to be some problem in wxWidgets code -- maybe the dialog is being disposed of too early or something is being freed too soon?

`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [99508]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.08.04.51 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-03-11 16:32:14.4398 -0400
OS Version:            macOS 14.3.1 (23D60)
Report Version:        12
Anonymous UUID:        4DFE2DCA-87E9-68E0-4DC2-A972C1229E8D

Sleep/Wake UUID:       05F031FF-9F44-44F2-B61B-FFC07E81CD83

Time Awake Since Boot: 390000 seconds
Time Since Wake:       28371 seconds

System Integrity Protection: enabled

Crashed Thread:        32

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x000000018d5b5af4

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [99508]

Application Specific Information:
abort() called

Thread 0:: bambustu_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x18d66b62c __ulock_wait + 8
1   libsystem_pthread.dylib       	       0x18d6ac49c _pthread_join + 608
2   BambuStudio                   	       0x103cd1d10 boost::thread::join_noexcept() + 188
3   BambuStudio                   	       0x1039b49a0 Slic3r::GUI::Job::join(int) + 116
4   BambuStudio                   	       0x103a90404 Slic3r::GUI::SelectMachineDialog::prepare_mode(bool) + 384
5   BambuStudio                   	       0x103a94ff4 Slic3r::GUI::SelectMachineDialog::show_status(Slic3r::GUI::PrintDialogStatus, std::__1::vector<wxString, std::__1::allocator<wxString>>) + 6704
6   BambuStudio                   	       0x103aa0b4c Slic3r::GUI::SelectMachineDialog::Show(bool) + 48
7   BambuStudio                   	       0x1055bcd30 wxDialog::EndModal(int) + 68
8   BambuStudio                   	       0x1036bbc5c Slic3r::GUI::Plater::print_job_finished(wxCommandEvent&) + 140
9   BambuStudio                   	       0x1056fdfb8 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
10  BambuStudio                   	       0x1056fdc98 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
11  BambuStudio                   	       0x1056fdabc wxEvtHandler::ProcessEvent(wxEvent&) + 56
12  BambuStudio                   	       0x1056fd90c wxEvtHandler::ProcessPendingEvents() + 236
13  BambuStudio                   	       0x105661774 wxAppConsoleBase::ProcessPendingEvents() + 112
14  BambuStudio                   	       0x1057067c8 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 60
15  CoreFoundation                	       0x18d786d80 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
16  CoreFoundation                	       0x18d786c6c __CFRunLoopDoObservers + 532
17  CoreFoundation                	       0x18d78629c __CFRunLoopRun + 776
18  CoreFoundation                	       0x18d78593c CFRunLoopRunSpecific + 608
19  HIToolbox                     	       0x197d4e448 RunCurrentEventLoopInMode + 292
20  HIToolbox                     	       0x197d4e100 ReceiveNextEventCommon + 260
21  HIToolbox                     	       0x197d4dfdc _BlockUntilNextEventMatchingListInModeWithFilter + 76
22  AppKit                        	       0x190f64ed0 _DPSNextEvent + 660
23  AppKit                        	       0x19174feec -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
24  BambuStudio                   	       0x10562da7c wxGUIEventLoop::DoDispatchTimeout(unsigned long) + 180
25  BambuStudio                   	       0x105706cd0 wxCFEventLoop::DispatchTimeout(unsigned long) + 44
26  BambuStudio                   	       0x105706df0 wxCFEventLoop::OSXDoRun() + 172
27  BambuStudio                   	       0x10562df4c wxModalEventLoop::OSXDoRun() + 88
28  BambuStudio                   	       0x105706eac wxCFEventLoop::DoRun() + 40
29  BambuStudio                   	       0x105686178 wxEventLoopBase::Run() + 132
[... 946 lines cut ...]
      "experimentId" : "65d62240cb0e2417d11ce6e1",
      "deploymentId" : 400000001
    }
  ]
}
}

Model: iMac21,1, BootROM 10151.81.1, proc 8:4:4 processors, 8 GB, SMC 
Graphics: Apple M1, Apple M1, Built-In
Display: iMac, Retina 4.5K (4480 x 2520), Main, MirrorOff, Online
Memory Module: LPDDR4, Micron
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4378), wl0: Aug 26 2023 18:07:59 version 18.20.439.0.7.8.163 FWID 01-4dd09646
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
USB Device: USB31Bus
USB Device: VLI Product String
USB Device: USB31Bus
USB Device: USB3.0 Hub
USB Device: USB2.0 Hub
USB Device: USB31Bus
Thunderbolt Bus: iMac, Apple Inc.
Thunderbolt Bus: iMac, Apple Inc.
`

blake-haas, 2024-05-02:
Experiencing the same issue. Crashes when sending to the cloud to print:
1.8.4.51