bambulab/BambuStudio #3548: Application keeps crashing or cancels when slicing is at 50%

garedgar, 2024-02-06:
### Bambu Studio Version

v01.08.04.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

Image Below:
[image]

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Download STL file from [https://www.printables.com/model/205674-nasa-chainmail-4x4-8x8-12x16-and-32x34/files](url)
2. Downloaded the NASA Fabric 12x16.stl option
3. Open Bambu Lab studio
4. Use default settings from printer.
5. Press on slice plate.
6. Process goes up to 50% Mark then it either just crashes and closes the application, no error  whatsoever. Or it clears the project on the application and returns to the home page as if the application launched from startup.
7. Tried this multiple times and it always crashes on the 50% mark, when i believe its trying to calculate supports, although the supports is disabled on the print.

### Actual results

1. Bambu Studio Closes
2. Bambu Studio clears current project and reboots the application the the starting screen.

### Expected results

To finish slicing plate and be able to send the file to the printer to start printing.

### Project file & Debug log uploads

[BambuStudio_Logs.zip](https://github.com/bambulab/BambuStudio/files/14182572/BambuStudio_Logs.zip)
[BambuLab_CrashProject.zip](https://github.com/bambulab/BambuStudio/files/14182573/BambuLab_CrashProject.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MetalRush, 2024-04-21:
Is there ANY progress on this bug?
We still encounter this problem, therefor cannot print certain items.

masystems, 2024-05-08:
I'm having the exact same issue, mine crashes at 15% sometimes too, i guess depending on options selected? It's very annoying.

QingZhangBambu, 2024-06-14:
https://github.com/bambulab/BambuStudio/issues/3318
Hi,  These methods may help. The CPU may not be stable enough to support multithreading. and whether it has functions such as automatic overclocking enabled?

MetalRush, 2024-06-14:
> #3318 Hi, These methods may help. The CPU may not be stable enough to support multithreading. and whether it has functions such as automatic overclocking enabled?

Thank you, QuigZhangBambu. That information helped me so much. I tested the "core exclusion trick" and ended up installing CoreDirector. This fixed my slicing problem. Thank you again, much appreciated.