bambulab/BambuStudio #3552: Ghost paths from upper layers create very inneficient paths for simple fills

ScottFerg56, 2024-02-07:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Microsoft Windows 10 Pro Version 10.0.19045

### Additional system information

Intel(R) Core(TM) i7-6700K CPU @ 4.00GHz, 4001 Mhz, 4 Core(s), 8 Logical Processor(s)
Installed Physical Memory (RAM)	32.0 GB

### Printer

Bambu Lab X1-Carbon Combo

### How to reproduce

1. Load project file
2. Slice
3. Observe tool path anomalies in layers 4, 5 an 6 (among others)

### Actual results

This is an image of the loaded project, a Starbucks logo. The objects create multiple levels, each a different color, each 2 layers thick and stacked upon one another so they do not overlap in the Z dimension. They do "touch" in that the top of one layer has the exact same Z value as the bottom of the next higher one. But I've discovered that adding a small gap does not eliminate the problem.

[image]

Layer 3 is the 1st layer of the 2nd level which is a white circle. The fill is nice and clean.

[image]

Layer 4 is the 2nd layer of the white circle which shows some odd anomalous paths that appear to be influenced by the black crown which doesn't appear until layer 9. There are also additional circular paths on the edge, perhaps influenced by the next green layer above. This tool path gets the job done, but I see no reason this layer should not be a simple circle and fill, which would be faster and more efficient. This path also introduces more possibilities for filament drag across the surface and other ripples which could harm the appearance of later layers.

[image]

Layer 5 is the 1st layer of the 3rd level which is a green circle. The crown anomalies persist but are not fully filled.

[image]

Layer 6 is the 2nd layer of the 3rd level green circle. It's filled fairly well, but again anomalies influenced perhaps by objects 3 layers up have made the fill more complicated than necessary. The lettering from the next white layer has also created very inefficient ghost paths at this level.

[image]

Layer 7 looks perfectly fine. The ghosts from level 9 that seemed to have effected the lower levels have not touched layer 7. Level 8 also shows some ghosting, but these images are enough to demonstrate the issue.

[image]

### Expected results

While the extraneous paths shown above did not noticeably harm the appearance of the end result in this simple case, I would expect to see much cleaner circular fills as shown in layer 7. In many cases I would expect the correction of this issue to result in faster and higher quality prints!

### Project file & Debug log uploads

[Starbucks_Coffee.zip](https://github.com/bambulab/BambuStudio/files/14189273/Starbucks_Coffee.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

rizwansarwar, 2024-03-01:
This is also the case when you have some text on top. It will leave these crazy ghost paths on top surface causing lot of surface ghosting. This is especially visible in shiny and dark colors. For example with black it should really be a smooth top surface, but it leaves shinny and matt areas because toolhead has to constantly fill around ghost paths with start/stop/restart path moves.

There should an option to fill the entire layer or part of it with a modifier for top surfaces.

XunZhangBambu, 2024-03-05:
Same as  #2906

XunZhangBambu, 2024-03-05:
@ScottFerg56  This issue arises due to the parameters bottom_shell_layers and top_shell_layer. These parameters, designed to enhance the strength of the top and bottom layers, cause a slight infiltration of layers in the top and bottom shapes.This layer will be printed as solid infill,resulting in the original flat layer being divided into multiple regions.

ScottFerg56, 2024-03-05:
@XunZhangBambu Are you suggesting that 'ghosts' appearing 5 layers below the influencing layer -- and through an unaffected layer -- could possibly be of any structural value? Or are you saying that these settings are the cause of the problem that needs to be fixed?