bambulab/BambuStudio #3554: Cutting An Assembly Does Not Preserve Parts Like It Did In v1.7

MadMax389, 2024-02-07:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab github releases

### OS version

Win 11

### Additional system information

10th gen Core i5. 32GB

### Printer

X1C, P1P, A1 Mini

### How to reproduce

Many times I create large models with multiple parts that I then cut into objects using the cut tool. In version 1.7 (and in Prusa Slicer), cutting the assembly (into objects) preserves the individual part structure within the assembly. So for example (in version 1.7), if I have a model with two parts and I cut it in half, Studio generates two new assemblies with two parts each (4 total stl's). In version 1.8, cutting an assembly seems to boolean-join the parts into only 2 stl's with no part assembly. This is causing me major problems with model design. Please take a look at this and restore this previous functionality!

To recreate this, 

1. Add a primitive to the plate and cut it vertically into parts. (Or drag a mutli-part model onto the plate).
[image]

2. You now have a multipart assembly
[image]

3. Now cut the assembly into objects (not parts) horizontally so it bisects both parts of the assembly.
[image]

### Actual results

Studio v1.8 now creates two new unassembled stl's. The left and right halves of the cylinder have been merged into the top and bottom halves of the cylinder.
[image]

### Expected results

In version 1.7 of Studio, the cylinder is cut into two assemblies, each with a left and right half, preserving any material and color assignments
[image]
Prusa Slicer also works as expected.
[image]

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-02-19:
Thank you for your question. I understand what you mean and we will investigate the cause. Thank you

Haidiye00, 2024-03-06:
I have fixed this issue, please wait for version V1.9. Thank you for your discovery and support~

MadMax389, 2024-03-06:
Excellent! Thank you. When can we expect v1.9?

Haidiye00, 2024-03-06:
Sorry, I can't make a decision on this. Just follow GitHub updates or the official website to receive notifications as soon as possible

Haidiye00, 2024-05-06:
The latest version has fixed this issue, please try again. thanks

closed 2024-05-06 by Haidiye00

MadMax389, 2024-05-06:
Yes, confirmed working now, thank you. Is #3876 a related issue?

Haidiye00, 2024-05-06:
I'm not sure if it's a related issue at the moment. Let me take a look at the reason first