bambulab/BambuStudio #3556: Default printer and filament

mbillros, 2024-02-07:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

mac big sur 11.7.10

### Printer

p1s combo

### How to reproduce

only defult maschine in printer setting, only crealite cr-6 shows not bambu lab p1s, all filament is gone cant shows nothing, all was working befor now its gone, have reinstall bambu studios many time nothing helps

### Actual results

nothing

### Expected results

i dont understand

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/14201899/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

XunZhangBambu, 2024-02-18:
@mbillros  Hi, i didn't reproduce your issue. The log you provided is network logs and i cannot see your operations in the log. Did you select P1S on the printer page？ A log like this may help
[image]

XunZhangBambu, 2024-06-26:
No responese for a long time, i think it can be closed now

closed 2024-06-26 by XunZhangBambu