bambulab/BambuStudio #3563: Wrong AMS Slot used when print started through app

marcel1099, 2024-02-08:
### Bambu Studio Version

1.8.3.51

### Where is the application from?

Bambu Lab Official website

### OS version

IOS 17

### Printer

Bambulab x1c

### How to reproduce

1. Upload i sliced model in a defined color 2. start that print through app but choose another AMS slot (which has another color)  

### Actual results

the printer loads not the chosen AMS slot but the one with the color the model was sliced with.

When the model was sliced with a color that isnt in the AMS, the printer gives you the warning that it couldn load the filament in the ams and you can cut it straight and retry.

### Expected results

The printer should use the ams slot selected in the app, regardless of what the original color it was when it was sliced.
(Like it works when using the touch screen directly on the printer)

On the touchscreen it even works when the color it was sliced with isnt in the ams.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/14214322/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-02-14:
```
1.Upload i sliced model in a defined color 
2. start that print through app but choose another AMS slot (which has another color)
```

@marcel1099 do you mean upload the file to makerworld?
if yes, could you show the link here

it seems an app related issue

marcel1099, 2024-02-14:
Hey,no i mean upload to the sd card thats in the printer :)