bambulab/BambuStudio #3564: Unable to set SKEW

BlackNet, 2024-02-09:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

win 11

### Printer

X1C

### How to reproduce

Functional prints are undersized.

### Actual results

Making functional prints yields smaller-than-expected dimensions. Using the hole/contour correction does *NOT* fix the issue.  The real problem is the printer is SKEWED and we are unable to correct for it.

### Expected results

Dimensionally accurate prints.

### Project file & Debug log uploads

any file produces this result.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

amzaldua, 2024-02-12:
By using the profiles labeled as 'high quality,' I have significantly mitigated this issue.

BlackNet, 2024-02-12:
> By using the profiles labeled as 'high quality,' I have significantly mitigated this issue.

it's not mitigated, the problem is still there.  The issue is X and Y, the 'high quality' is more about Z.

Thibo-do-it, 2024-02-13:
I second this request! It would be nice to edit the skew in the firmware to correct as shown in the link below

[Marlin skew correction](https://marlinfw.org/docs/configuration/configuration.html#bed-skew-compensation)

SaltWei, 2024-02-18:
@BlackNet Could you please raise a tickets on Bambu Handy app? Then CS and printer team will help to check whether there has problem with the x-y sliding rail first.

BlackNet, 2024-02-18:
> @BlackNet Could you please raise a ticket on Bambu Handy app? Then CS and printer team will help to check whether there has problem with the x-y sliding rail first.

By all rights, this is not a mobile app problem. This is a core functionality problem with ALL printers with a very easy simple solution that has been in use for an easy 50 years.

SaltWei, 2024-02-19:
> By all rights, this is not a mobile app problem.

Of course, not app issue. I mean raising a ticket by Bambu Handy app and printer team will evaluate further.
https://wiki.bambulab.com/en/x1/troubleshooting/upload-log

I have also sent this github issue to them to evaluate as well. Thanks for suggestions!

BlackNet, 2024-02-19:
Why would it be required to upload log files for a skew correction issue?  The only good that will do is tie up support resources with needless data and delays. 

It's very painfully obvious there are 0 skew corrections, and 0 provisions for skew in any manner from the firmware (this is via numerous forum posts on the subject with a large number of people trying various things to apply multiple methods which are tried and true).

Perhaps I am missing something here.  What would be in the logs that could help with skew?

PanamonCreel, 2024-02-19:
@BlackNet 
Take a deep breath.  A factory build rigid frame core X-Y printer like your X1C is unlikely to have a significant amount of Skew unless there is something wrong with your printer.  
I would assume that support staff here just wants to get as much information as possible, including the log file, to get a better picture of what the root cause of your issue could be.  
The original problem description you provided is rather meager and from my 3rd person viewpoint and experience "functional prints are undersized"  does not equal "skewed printer". 
Maybe you've made some test prints to verify the squareness of the XYZ axis to each other and if so then you should share the measurements thereof.

MrDB42, 2024-02-23:
Right now, the only way to correct any XY skew is "mechanically". I was suffering from some XY skew that was also affecting some of my prints. I stumbled across this YouTube video which isn't labeled as XY skew correction (banding issues) BUT following it did correct my skew significantly. 

https://youtu.be/izdBZ9IQfqI?si=IVtB_ps9ufiLTqO_

He suggests using layers of tape for the adjustment shim... I disagree and use feeler gauges to measure what is required to make the triangles line up. Then I just cut off some of the feeler gauges and used double stick tape to secure them (remember to take into account the thickness of the double stick tape).

Yes, I tested with printed object, a rather LARGE (the bigger the better for comparisons), thick (for rigidity) hexagon, with lots of chamfered edges (so the caliper faces weren't hitting sharp corners) works perfectly for this. Measured the object to verify the amount of skew, and to make sure the X and Y distances matched up. Prior to the modification, yeah... things were off. X and Y distances didn't match (±0.05), neither did the diagonals (±0.10). But after the modification, I was down to ±0.01 on all my measurements.

As for "A factory build rigid frame core X-Y printer like your X1C is unlikely to have a significant amount of Skew", sorry but nothing is perfect "out of the box". There are always tolerances and slop in assembly (built on a Monday or Friday, or someone having a "bad" day, etc.).

Personally, I would prefer the software method of XY Skew compensation, but "you do what you gotta do" in the end. I might try again to see if I chase it down to zero hundredth on my calipers.

kbhse, 2024-03-29:
[See this post](https://www.reddit.com/r/BambuLab/comments/1adh0bn/comment/kwp3rh3) for a way to compensate for Skew in firmware.

GuzziRaz, 2024-04-02:
So it was supported all the time and no-one knew about it. This is a good example of how ridiculous it is that we haven't got a G-Code reference for Bambu printers. We need that.

MrDB42, 2024-04-03:
I did the software skew correction, M1005 X(value) Y(value), and I was able to get it diagonal difference down to 0.00 via good quality digital calipers.

I didn't use the model that Bambu supplied because it was just to flimsy and compressible. I created a nice chamfered, rounded corner, with thick walls, hexagon in Fusion. By measuring the exact same position and with my best ability same amount of pressure on the thumb wheel.

After I did a full calibration via the screen on my X1C. Boy are my prints nice now. Also, I swear after doing the motor noise cancelation, the printer got even quieter than before (subjective, I know).

Now if we can get the G-codes for steps and backlash correction life would be complete for printing functional prints.

Before I bought my X1C I did email pre-sale support I did ask about auto step and backlash calibration. They assured me that the printer does it... But now I'm not too sure since I still have inaccurate dimensions with my prints.