bambulab/BambuStudio #3566: Select infill patern for bridge surface

M-Spit, 2024-02-09:
**Is your feature request related to a problem? Please describe.**
The bridge infill pattern is always rectalinniar, which is fine when you print square parts.
However, I try to print a round part (a cillinder) with a widening of said cilinder in layer 150. Instead of the rectaliniar infill, I'd like to use concentric or even archymedian spiral patterns.
Especially when trying to print waterproof parts, concentric is worse to deal with (even with a sealant, in certain directions some gaps between the lines will leak)
It will also add to printing speeds for these layers.

**Describe the solution you'd like**
I'd  like a setting in the bridge settings where you can choose the surface infill pattern of a bridge.

**Describe alternatives you've considered**
Manually ironing the surface of the bridge, which is labor intensive.

**Additional context**
[image]
The blue bridge would be a lot nicer when printed as a concentric infill or a spiral.
https://www.reddit.com/r/BambuLab/comments/15tfkrb/how_to_change_the_bridge_pattern/
A reddit post describing the same issue

Thanks in advance!

BotSpace, 2024-03-05:
I second this, would be very helpful for my project.

riccalisci, 2024-10-24:
please include this feature! i am sure it will be very helpful

n3dwards, 2025-03-31:
**Throwing my my support for this here, along with some context/suggestions for how this would work.**

I've just started using PLA to support PETG and it works so well that I can now add supports for floating flat surfaces that have zero z-distance between them and the support material. This functionally allows me to have cleanly printed bottom surfaces that are not directly on the build plate. However, I have no options for configuring the pattern since these lines are considered "bridge" and not "bottom surface".

I presume the bridge line type doesn't support patterns because normally any pattern other than linear would not really bridge properly in a normal bridging scenario. However, with 100% support underneath (or close to it), they are not actually bridging any gaps at all.

**My suggestion:** change this so that if "bridge" lines are fully supported, the slicer will stop treating them as bridges and switch to "bottom surface", which would allow for pattern selection.

chadwallacehart, 2025-08-30:
+1 for this feature. I also have a project where I could also use a non-linear pattern for bridges