bambulab/BambuStudio #3569: Failed to connect to the printer

rexmorgan99, 2024-02-10:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab x1 carbon

### How to reproduce

Go to Bambu Studio click the Device tab along the top. 

### Actual results

[image]

### Expected results

I expect the software to connect to the printer. Thereby allowing me so send files to the printer. 

### Project file & Debug log uploads

I have tried dragging and dropping a file here. However I cannot do that. When I try nothing happens. 
[no_project.zip](https://github.com/bambulab/BambuStudio/files/14230616/no_project.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

lanewei120, 2024-02-14:
@rexmorgan99 
could you also help to upload the log files here

you can follow the guide to find the log files
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

sonjamichelle, 2024-02-14:
I am getting the same errors and having the same issue here.
[log.zip](https://github.com/bambulab/BambuStudio/files/14284677/log.zip)

[Boxes for small parts with cover - 5455700.zip](https://github.com/bambulab/BambuStudio/files/14284683/Boxes.for.small.parts.with.cover.-.5455700.zip)

walterwongbbl, 2024-02-17:
@rexmorgan99 
Can you check the network on the printer side? I checked the LOG you provided and found that the server did not return any data related to the printer. When did this question start.

rexmorgan99, 2024-02-18:
The network was just fine. I did restart the printer and was able to
connect.

walterwongbbl, 2024-02-18:
@rexmorgan99 
Can you connect to the printer properly now?