bambulab/BambuStudio #3570: Filament assignment in the AMS doesn't work

Keno55, 2024-02-11:
### Bambu Studio Version

1.8.451

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Intel(R) Core(TM) i5-9400 CPU @ 2.90GHz   2.90 GHz  64-Bit-Betriebssystem
RAM: 8 Gbyte

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Go to device
2. AMS
3. for example new filament Slot A3
4. edit any Slot with question mark and confirm
5. After 5.-7 Seconds the new filament disappears and the question mark comes back
6. When I do the same on the printer display it works
7

### Actual results

It doesn't work...only with the display on the printer

### Expected results

I want to be able to easily assign the filaments in the studio. Why they keep disappearing after confirmation remains a mystery to me

### Project file & Debug log uploads

It doesn't matter which project is loaded.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MadMax389, 2024-02-13:
I can confirm this behavior. For me the problem resolved when I assigned the custom filament from the printer screen. Once I did that for at least one filament, I was able to assign them all normally from Studio.

DanBao-Bambu, 2024-02-18:
Yes, this is currently the correct workflow for using custom consumables normally:
1. Ensure that the user's preset cloud synchronization is enabled
2. X1C model
3. After creating a custom Filament, go to the printer screen and enter the Filament settings interface to check if your custom Filament already exists. This ensures that the printer has been synchronized from the cloud to the newly created custom Filament.
4. Custom Filament can be set up normally on Studio.
I know that setting up the printer is quite complicated, and we are currently optimizing the logic to provide users with a better user experience.
For specific details, please refer to the following link:
https://wiki.bambulab.com/en/software/bambu-studio/custom-filament-issue

closed 2024-03-20 by bam-winterfell