bambulab/BambuStudio #3571: Camera lost connection

ReneZoetekouw, 2024-02-11:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS

### Printer

Bambu X1

### How to reproduce

Ofter camera x1 is doing well at first print. 2nd print camera stream is not commenting. No error message, just simple black screen. Also restart Bambu lab does not help. It remains loading and then switch back to black screen

### Actual results

Nothing. (Green start video comes available again)

### Expected results

Stream should start

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/14232145/log.zip)
[Naamloos.3mf.zip](https://github.com/bambulab/BambuStudio/files/14232146/Naamloos.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2024-02-18:
I found -90 error in the logs.
May some as https://github.com/bambulab/BambuStudio/issues/3199
Please reboot the printer and retry.

ReneZoetekouw, 2024-02-18:
Hello,

Indeed, after reboot of printer it works……. For some time and then same issue again. Seems to be a more structural issue.

Hope there will be a solution for this soon.

Rene

bambu123, 2024-02-19:
> Hello, Indeed, after reboot of printer it works……. For some time and then same issue again. Seems to be a more structural issue. Hope there will be a solution for this soon. Rene
> […](#)

Yes, there will be a fix for X1, version 1.8.0

ReneZoetekouw, 2024-02-20:
And when is this fix expected?

bambu123, 2024-02-21:
> And when is this fix expected ?

About three months for one version