bambulab/BambuStudio #3572: Bambu Studio does not  connect to server

Honda16vtec, 2024-02-11:
### Bambu Studio Version

Version 1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: 11th Gen Intel(R) Core(TM) i7-1185G7 @ 3.00GHz Memory: 32.0GB GPU: Intel(R) Iris(R) Xe Graphics

### Printer

Bambu X1 Carbon

### How to reproduce

1. Go to Slice Plate 2. Print plate 3. Error, Connecting to server
[image]

### Actual results

[image]

### Expected results

I should be able to select SEND
[image]

### Project file & Debug log uploads

Sun Feb 11 09:58:25:test bambulab start...
Sun Feb 11 09:58:25:[test_bambulab]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?name=slicer&version=01.08.04.51&guide_version=00.00.00.01
Sun Feb 11 09:58:26:test bambulab ip resolved = 104.18.2.216
Sun Feb 11 09:58:26:test bambulab ok

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-02-14:
@Honda16vtec could you help to upload the log here?
just following the guide https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

sonjamichelle, 2024-02-14:
I could have sworn I posted a comment:

Here it is again, seeing the same behavior here too:

[image]

Logs:
[log.zip](https://github.com/bambulab/BambuStudio/files/14289495/log.zip)

Project:

[Boxes for small parts with cover - 5455700.zip](https://github.com/bambulab/BambuStudio/files/14289497/Boxes.for.small.parts.with.cover.-.5455700.zip)

sonjamichelle, 2024-02-15:
Even when it DOES connect, the send times out.

[image]

[image]

[image]

[image]

[image]

[image]

I had turned my printer off about two weeks ago. I had another pressing project to work on and no real stock of filament on hand anyhow.

I turned it on the other day, opened BambuStudio, it asked for an upgrade, I allowed it, the network plugin as well.

Now this!

I even tried rolling back a version after uninstalling an dmaking sure all remians were off the computer.

Bambu won't talk to me since I bought the printer thorugh Matter Hackers hee in the states. Matter Hackers has YET to return my email, calls or support requests. It is now Thursday, I put the request in Monday morning.

Everything ELSE on my network is working as exepected. I have restarted, rebooted, unplugged, plugged every bit of networking gear I have, from the modem, router, switches to the Printer. Nothing has resolved the situation.

I have tried different WiFi settings, names, channel, even different hardware, my phone even!

ALL was well 2 weeks ago when I had 1.8.3 installed. I was a printing fool, cranking out prints all day and night. It would only take about 10-30 seconds to transfer a file depending on size and complexity.

Now it won't transer even the simplest file or maintain a connection either.

It has been about 5 minutes now and I have not passed 60% on the transfer.

sonjamichelle, 2024-02-16:
Nothing? No comments? IS this a bug that is being worked on? Is there somethignwith the firmware on the machine?

SOME form of feedback would be appreciated!

walterwongbbl, 2024-02-17:
@sonjamichelle 
Is there any error message on the printer side?

sonjamichelle, 2024-02-17:
NO!

Just the wifi keeps turning off. No erros from teh printer, self test passes and calibration passes. Only message I get is that it can't connect to the router on the internet. YET EVERYTHING ELSE on the network is operating as normal as usual.  The fact that I've been getting 2000+ ms pings leads MatterHAckers to check the hardware. Though it stumps them as to why after the printer sitting POWERED OFF for two weeks since my last print that this is happening is confusing to them, but not unheard of.

I finally had contact from MatterHackers, the company I purchased the printer from. They are sending out a new mainboard and WiFi  module.  I'll see if it is a hardware issue like they may have suggested. They also said there was no easy way to roll back a firmware update either. Hence the new parts.

walterwongbbl, 2024-02-18:
HI @sonjamichelle 
If the average latency exceeds 400, the printing will not work well. That's strange. Have you tried restoring the factory settings?

sonjamichelle, 2024-02-19:
Tried it, dit NOT work!

WiFi connection CONTINUES to drop! Even though EVERY OTHER F-ING WiFi device on the network STAYS connected and performs as expected.

Printer will NOT maintain a connection, as I said BEFORE, tried phone wifi, router wifi, mesh wifi, 6 differnt AP settings and forcing to 2.4GHz.

NOTHING resolves the issue. Either the "new" network plugin introduced new bugs across the board, or the WiFi board is bad. Matter Hackers is sending repacement parts Monday.

IF THAT does not resolve the problem you will have to dig deeper.

I don't know what more I can add, I've already added screenshots and IIRC 24 logs. With the network logs either being encoded or encrypted I can NOT open them to debug. Which BTW is a right pain in the a--.

walterwongbbl, 2024-02-21:
@sonjamichelle 
Waiting for you to replace the parts

closed 2024-03-27 by walterwongbbl