bambulab/BambuStudio #3575: Sending print over the lan failed (Err -4020)

ReverantGR, 2024-02-12:
### Bambu Studio Version

v01.08.04.51_20240117

### Where is the application from?

Bambu Lab github releases

### OS version

OpenSUSE 15.4

### Additional system information

AMD 5950x
64GB RAM
AMD RX6600XT

### Printer

Bambu Lab X1E (Ethernet, LAN Mode) Fw 01.00.02.00

### How to reproduce

Open Bambu Studio, open any project, slice, click Print Plate, click Send.

### Actual results

Sending print job over LAN failed.
Error: -4020
Error: Failed to upload print file to FTP. Please the network status and try again.
Extra info: [ftp code]: 9
[reason]: Access denied to remote resource: Server denied you to change to the given directory. 

### Expected results

  Printer should proceed with the print and not deny. Works fine on Windows on v1.8.2.56.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/14236431/log.zip)
[0.20mm-Bambu-PLA-Profile.zip](https://github.com/bambulab/BambuStudio/files/14236454/0.20mm-Bambu-PLA-Profile.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2024-02-17:
@ReverantGR 
4020 indicates an attempt to establish a connection with the printer, but failed. Usually, this type of failure is related to the network. You can try using your computer to ping the printer's IP and check if it is connected. Also, please provide the log of Studio.
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

elvenwhite, 2024-02-25:
Same symptom here. Able to ping the printer, also able to connect it via FTP. 
I was able to Send, but unable to Print plate.

Tried power cycling, turning Lan only mode off then on, and resetting access code. No success.

walterwongbbl, 2024-03-02:
Hi. @elvenwhite 
May I ask if you can provide Studio's LOG logs? I will help you analyze the issue.

elvenwhite, 2024-03-02:
[bambulab.zip](https://github.com/bambulab/BambuStudio/files/14470563/bambulab.zip)
Hope this helps. I suspect there's something with character set, because the problem occurs with some specific 3MF only.

LayerCakeMakes, 2024-03-09:
I didn't get the logs at the time but this just happened to me yesterday again too on my X1C.
[image]

Viewing the local lan video stream through bambu studio worked fine but the upload always failed.

So far this has happened twice too me and the only way to get it working again is doing a printjob through the cloud and then switching to LAN mode again.

I also noticed that the status website linked in the error message showed that the "Printer Services" -> "Binding" was experiencing issues this time, I hope this is not related as everything should be local ... but maybe it isn't.

LayerCakeMakes, 2024-03-09:
It seems to happen for me with this print profile for example: https://makerworld.com/en/models/59601#profileId-62387
I just opened it in Bambu Studio switched to X1C as the printer and 0.2 Standard profile.

walterwongbbl, 2024-03-11:
@LayerCakeMakes 
We will try to reproduce and fix this issue

serpent213, 2024-04-07:
Just stumbled upon the same thing. I think it happened when I used Unicode in the Fusion project title (a “–” in this case) and then exported (“3D Print”) the design as 3MF project (even though the 3MF filename did NOT contain Unicode later on).

eolder, 2024-04-15:
I've experienced this with many models from Makerworld. I cannot reproduce with the above linked profile, so I've included an example that causes this for me - https://makerworld.com/en/models/174561#profileId-191936

Element115sw, 2024-04-15:
Dito. This model https://makerworld.com/en/models/239797?from=search#profileId-263097 after changing nozzle from 0.4mm to 0.2mm

MostHated, 2024-04-21:
I am getting this `-4020` error when trying to print from lan with the following: https://makerworld.com/en/models/162780#profileId-215962

It is not network related, I am printing something else right now as we speak. There are a few project downloads I have done which do this, and exporting to stl worked, but this particular project is 9 plates and setup quite specifically. 
I would much rather figure out what the actual issue is so things work as intended, so I don't have to try and recreate the entire setup again from scratch via exporting dozens of items to stl's.

AngelDLM08, 2024-04-29:
Got FTP error -4020 but two different codes: 
28: Timeout was reached: FTP response timeout
and after closing and opening OrcaSlicer
25: Upload failed (at start/before it took off): Failed FTP upload: 533

walterwongbbl, 2024-04-30:
We have reproduced this issue, which seems to be related to the character set. We will fix this issue.

walterwongbbl, 2024-04-30:
If the model file comes from Maker World, it will enforce the use of names from Maker World when sending prints. However, if the name contains special characters, it can cause the sending to fail. We will take some measures, such as filtering out special characters from the name.

probonopd, 2024-05-09:
Same issue trying to print https://makerworld.com/en/models/206463?from=search#profileId-226622 using Bambu Studio on A1 mini with factory firmware (not updated because updating is apparently not possible in LAN mode)

[image]

AngelDLM08, 2024-05-09:
> I suspect it something with bambu network plugin that OrcaSlicer installs from bambu since the only way I was able to sent print via wifi is reboot printer, disable network plugin, reboot OrcaSlicer, enable network plugin and only then send print file

Was wrong about that, just coincidence, find a way - connecting through another subnet and all works like a charm. Seems after updating X1C to 01.07.03.00 it suffers from unclear for me network issue

AngelDLM08, 2024-05-09:
> Same issue trying to print https://makerworld.com/en/models/206463?from=search#profileId-226622 using Bambu Studio on Windows 11 (German) and A1 mini with factory firmware (not updated because updating is apparently not possible in LAN mode)
> 
> ![image](https://private-user-images.githubusercontent.com/2480569/329370542-db7cf1b3-2b52-4f68-899d-827c8d298456.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUyODUxMzQsIm5iZiI6MTcxNTI4NDgzNCwicGF0aCI6Ii8yNDgwNTY5LzMyOTM3MDU0Mi1kYjdjZjFiMy0yYjUyLTRmNjgtODk5ZC04MjdjOGQyOTg0NTYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MDlUMjAwMDM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzNjNjEzMjBjODZmZjVkYWFkMmM0Y2FjYjMyMTdkOGE1NTEwYjU5ZmViOWVjZjY3NGFmNDhmM2U3ODk0MDlkMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.7ghfVR5h7XsSbjlXcNJ4ZurrFS8e9vz3U4d_PmM1kPA)
> 
> Editing the name to just "honey" did not help:
> 
> ![image](https://private-user-images.githubusercontent.com/2480569/329371354-6b68284a-30f1-4149-ba2e-cc109741024a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUyODUxMzQsIm5iZiI6MTcxNTI4NDgzNCwicGF0aCI6Ii8yNDgwNTY5LzMyOTM3MTM1NC02YjY4Mjg0YS0zMGYxLTQxNDktYmEyZS1jYzEwOTc0MTAyNGEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MDlUMjAwMDM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmZlZTNjNWIwOGNmMzVjMjIzMWY2MzdhYWJkY2EwZTI3NjFiMWQxODAyZWI2YTQxNTBmNDI4YzI2NTQ3NWQ1OSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.zRLN2Ux9OydEk6J2qAyMsGj7VDTGnAk3q3MC92boFTQ)
> 
> Is there any workaround?

Try FileZilla to connect to FTP,

probonopd, 2024-05-09:
Other files can be printed. So it must be something about this 3mf file from Maker World.

AngelDLM08, 2024-05-09:
> Other files can be printed. So it must be something about this 3mf file from Maker World.

Try to print that using OrcaSlicer, fork of bambu studio

probonopd, 2024-05-09:
Just tried that, fails in the same way. Other files print.

walterwongbbl, 2024-05-13:
@probonopd @AngelDLM08 
We have fixed this issue and a new version will be released later.

probonopd, 2024-05-13:
Thanks, it works for me in the version [1.9.1 Public Official](https://github.com/bambulab/BambuStudio/releases/tag/v01.09.01.66).

eolder, 2024-05-13:
Thanks @walterwongbbl and team.

eolder, 2024-05-13:
looks like it was this commit, 1bcf30c, for anyone who was curious :)

AngelDLM08, 2024-05-13:
> Recently updated firmware on X1C from stock one, and problem starts... Used only in LAN-only before and got no issues at all. Now I have FTP errors -4020 but two different codes: 28: Timeout was reached: FTP response timeout and after closing and opening OrcaSlicer 25: Upload failed (at start/before it took off): Failed FTP upload: 533
> 
> I suspect it something with bambu network plugin that OrcaSlicer installs from bambu since the only way I was able to sent print via wifi is reboot printer, disable network plugin, reboot OrcaSlicer, enable network plugin and only then send print file

I found solution for my problem - disable windows firewall, IDK why  that matter since with A1mini works fine. Got nothing with that issue, but still can be relevant for someone

JW97, 2024-08-20:
On a similar note, I've also been struggling with FTP -4020 errors on a corporate network (after lots of issues getting discovery functioning in our network config).
We have prints get through to the printer on occasion, but quite often we get 'FTP Timeout errors'.

The file makes it to the printer, as removing and re-inserting the SD Card to trigger a refresh pulls in the file.

Very annoying, and not what we expected purchasing this machine.

Does anyone, or Bamnu themselves have any suggestions?

tudi55, 2024-09-15:
If you use SolidWorks, exporting as AP203 instead of AP214 solved the problem for me in some cases.

houseofbugs, 2024-09-18:
Tried all the things in the suggestions here. Still having this issue with the ftp error 9. Really sad that a $1400 machine fails at printing over the network vs my crappy ender machines with octoprint..

walterwongbbl, 2024-10-17:
Hi @houseofbugs 
We have found too many issues with FTP, so we have decided to re implement the file sending function.

skintigh, 2024-11-28:
> We have reproduced this issue, which seems to be related to the character set. We will fix this issue.

When???

skintigh, 2024-12-02:
Well obviously the answer is never, but it seems their printer attempts to open a new connection back to the PC and to upload a file containing nothing other than "this is a test file." So in my case disabling the firewall allowed me to print again. This was despite the fact I had already allowed their software to bypass the firewall, and that this worked fine for a while, until one of their software updates broke it and their tech support decided to go on a 9 month vacation.

Icemic, 2024-12-04:
The problem is located at the "project info" in your .3mf.
In the title, there's a slash which is invalid for a filename. I think the uploader use the title as (part of) filename directly instead of a uri encode, the invalid path causes this error.

As a workaround, edit the title and remove the slash, reopen your project, anything is solved.

This bug has been in bambustudio or orcaslicer more than a year...

[image]

tudi55, 2024-12-07:
Please for the name of god fix this problem already. It's making my experience with a otherwise printer filled with frustration.

houseofbugs, 2024-12-08:
> Please for the name of god fix this problem already. It's making my experience with a otherwise printer filled with frustration.

If you're tired of it and dont use multicolor the K1 and K1C from Creality have been solid. I have all those and the K1 MAX. They just work with Orca out of the box.

skintigh, 2024-12-09:
For me this problem has nothing to do with filenames, slashes, unicode, coddupted files. 100% sure.

*Almost* 100% of my prints fail when the firewall is on -- bambu gets to either 30% or 70% and either freezes and crashes, or reports an FTP error. If I quickly turn off the firewall when it is stuck at 30% or 70% the print succeeds almost 100% of the time. I have quintuple verified I'm letting bambu through the firewall, nothing is logged as blocked. But watching the traffic on Wire Shark shows the printer is CONSANTLY having diarrhea of packets. MQTT, tons of spurious retransmissions, tons of resets, tons of sequence packets, and this is while the printer is completely idle.

It's close to the router so that shouldn't be the issue. Sure do wish it had an ethernet port, or USB port like virtually every printer on the freaking planet. It seems the firmware is just a disaster of bad programming.

R-a-K-i, 2024-12-18:
I have the same problem with the Windows fiewall on. I'm using Windows 11 with BabmStudio Version 1.10.1.50

tudi55, 2024-12-24:
Ok. So I might have found the problem for myself. Firework in my case si irrelevant. I need to give the project file (not only the one you send to the printer, the one saved on your pc as well) a name that is just lowercase connected letters and no characters. And if it fails to change the name again. That seems to make it more consistent? Still can't belive this is still not fixed.

skintigh, 2024-12-26:
So I left a comment confirming this, it seems to be deleted. 

I tried "ring0.stl" and it would not print while renaming it "ringzero.stl" did print. However, other files with no numbers still didn't print. Seems there are multiple glitches, or this just triggers the glitch sometimes?

zbrozek, 2025-01-19:
If I use a standard FTP client to connect to the printer, the file appears to be successfully uploaded. If I select the file from the touchscreen, it prints. I just can't remotely start a job on an X1E in LAN mode because of FTP timeout errors.

skintigh, 2025-01-20:
I set up firewall logging, it shows no packets being dropped, even though disabling the firewall allowed prints to succeed.

When I viewed the log for Windows Filtering Platform, I found the dropped packets from port 2021. It seems BS expects a packet on port 2021 or else it times out. But it doesn't bother to listed to port 2021. So the printer sends a packet to port 2021, WFP sees it as an unsolicited port scan and drops the packet before it can ever get to the firewall.

Turning off the firewall also silently shuts down WFP, so it looks like turning off the firewall makes prints work when really it seems turning WFP off is what makes prints work.

All of this begs the question of why BS requires a packet to print but doesn't actively listen for it, and a how in the world does that even work when WFP is off? Is that something to do with passive FTP? Or is there a race condition that BS loses almost every time and opens the port too late? I don't get it.

I saw some python scripts that will spoof some of these packets and tell the printer who to talk to, but I don't know if they help this issue. 

ANYTHING AT ALL FROM BAMBU would be a welcome change. Even a single piece of info, a single reply, literally ANYTHING

maaaaaaaaaaaaaaaaaaaaaaaaaaax, 2025-01-26:
Thank you for pointing out the issue with WFP. I was able to resume printing by changing my WiFi connection to be on a private network instead of turning off the Firewall completely. We will probably be able to fix this issue by adding some specific firewall rule to the Windows Firewall? I will resume testing once my current print is done and I can try to send prints to the printer again.

skintigh, 2025-01-26:
Are you sure your firewall isn't off for private networks? There are 3 on/off switches for 3 flavors of network.

For me it doesn't matter if I'm on public or private, I tried both. I also recently replaced my wifi router and the issue persists so the router doesn't seem to be a cause.

As best I can tell, WFP has decided nobody is listening on port 2021 and is blocking the packets, and BS seems to require that packet for no apparent reason and fails without it. It must be listening to the port somehow, but it may have something to do with using the FTP protocol which has been obsolete for decades.

maaaaaaaaaaaaaaaaaaaaaaaaaaax, 2025-01-26:
Yes, it's an almost brand new laptop, I've also never changed any firewall settings before. I used the zodiacon WFPExplorer to check if any packages are dropped by the WFP and also checked the firewall logs, but there are no packets with port 2021 blocked. Also, I see packages to Port 2021 in Wireshark. I assume those get filtered out before they reach wireshark for you? If thats the case, can't you create a firewall rule to accept all packages that go to this specific port?

skintigh, 2025-01-27:
You enabled logging with the firewall, I assume? It defaults to logging nothing.

I have firewall rules, but it doesn't matter because it seems WFP eats packets before they even get to the firewall. The logs where they are eaten suggest as debugging you make sure someone is listening on the port, but maybe that's a red herring.

I should check out that explorer you mentioned.

maaaaaaaaaaaaaaaaaaaaaaaaaaax, 2025-01-27:
Yes, and I see packets being dropped in those logs, so it generally works. In my understanding the received packets go through WFP, then firewall, then wireshark. If I see them in Wireshark, WFP and the firewall let them through.
How did you find out about the dropped port 2021 packages? I didn't find an easy way to see the log of WFP, except with the WFPExplorer.

skintigh, 2025-01-28:
I used the instructions here, and it was definitely blocking some of the packets. I traced the rule to WFP. Using the tool you mentioned and testing more, I saw "callout terminated" messages and allow messages. Not sure what that means.

The only thing consistent is the way in which it's inconsistent -- if I mess with the firewall rules. or turn it off and on, or similar, the very next print goes through. The very next print after that fails, as do all subsequent prints. So the firewall (and WFP) can be on and exactly 1 and only 1 print will work. After that they will all fail, unless I turn off the firewall+WFP and then they work.

I have no idea what that means. Bambu botched sequence numbers somewhere?

melonexe, 2025-02-23:
Why this hasn't been fixed yet is beyond me. It's just basic functionality which should work.

I love their product but my god this is infuriating. My ender 5 printer with octoprint on a dreadful little NUC PC had better reliability than the LAN transfer in Bambu lab, HOW?!

Makes me want to not use this printer sometimes as I use it as a tool not a hobby. If a tool is broken and I can't fix it... It usually goes in the bin or gets sold.

None of the work arounds in this thread have worked for me and I don't want to use cloud because why should I have to rely on some random server halfway across the world to send a print to my printer downstairs?

Sorry for the rant but I find it disgusting Bambu doesn't seem to want to even try to fix this.

Thirtythree3d, 2025-03-10:
I suffered this exact same issue. Tried every fix suggested. In the end I worked it out to be the SD card. Formatted it and the issue went away. Maybe something to try if you haven't already.

dustindufault, 2025-03-15:
I'm in this boat too. Very very frustrating. It happens off and on for me. No rhyme or reason as to why, ever.

dustindufault, 2025-03-15:
Here's an interesting tidbit from my hours and hours of uncomp testing for Bambu --- even when in this state if I submit a calibration from the calibration tab, those prints start immediately. And, like others, every other connection to the printer works fine for me, the camera and print head / bed controls are all operational.

maaaaaaaaaaaaaaaaaaaaaaaaaaax, 2025-03-15:
@dustindufault Did you try the suggestion to format the SD card? I didn't have time to test this yet, but it will be the next thing I try.

dustindufault, 2025-03-16:
> [@dustindufault](https://github.com/dustindufault) Did you try the suggestion to format the SD card? I didn't have time to test this yet, but it will be the next thing I try.

Yes, I did. I tried formatting 3 different cards. The original one - the cheap little black one that came with the machine and was working, normally, and two different high speed, high quality, san disk ones. I tried formatting on the machine, on my desktop, formatting to a mac partition to force the machine to re-format everything, etc. No dice.

And - Bambu Studio seems aware of the sd card status instantly. If I remove the card for formatting or present one that isn't properly formatted, the send print window will immediately notify me that needs to be remedied.

I spent way too much time trying to figure out the problem yesterday and as far as I can tell there seems to be an issue with some of the threads running on the machine. I attempted to install the open source firmware to bypass the issue which requires ssh access. I set it up properly but when attempting to access it, it will time out on that too. So I tried to ssh in myself and it's readily apparent that the machine / processor are just bogged down for some reason. It would take several minutes for ssh to display a prompt and forget about actually using it! Meanwhile the display at the printer doesn't lag at all, so.....

I dunno. But theres a weird problem going on inside the machine itself.

skintigh, 2025-03-16:
I also noticed that tidbit you mention, and a few more:

- FTPing with any client I've tried works perfect. Any except BambuStudio
- Calibrations print instantly, as mentioned, when all other prints fail
- BS sends a file "verify_job" and sends the file you want to print. If you delete them and watch from a working FTP client, BS **successfully** transmits both files, THEN hangs at 70%.
- Even though I'm on a private network, if I turn off the _public_ network firewall the very next print and ONLY the very next will often/always(?) work, but all subsequent prints will fail. WTF? Oh and the log shows it blocking no packets.
- Even though my private network firewall is blocking no packets according to the log, turning it off allows prints to work, _suggesting_ Windows Filtering Platform is blocking spammed packets as a DOS attack, but who knows. I haven't actually tested if it lets just the next print work or many work, as I always turn my firewall back on.
- When hanging at 30% I can cancel the send, but if I press cancel when hanging at 70% in ANY VERSION of BS, BS locks up and crashes. Test much, Bambu?

I heard back from support. This is what they said, it wasn't exactly helpful:
---
Studio establishes a connection with the printer and sends a file of approximately 1K to verify if the local network function is ready [I assume they mean the file "verifiy_job"]. If the reception fails [**it doesn't**], this type of situation can occur. You can try deleting the configuration folder, which stores FTP information, to see if this resolves the issue.

Additionally, you can also attempt using version 1.9.7.56 [I would love to but this version **doesn't exist**. I tried 1.9.7.52 but it didn't seem to work for me.] This version has modifications in the communication logic with LAN mode compared to 1.10, which may help in determining if the issue can be resolved.

In future updates, we plan to optimize this aspect by utilizing different transmission methods and verification techniques. Please keep an eye on the Bambu Studio Update Release Note for further information. [So they are working on something for release sometime, that's something]

ffafe5, 2025-03-22:
@Noisyfox please apply some of your magic here!

melonexe, 2025-03-25:
I have now sold my Bambu printer because of this issue. I wish they could have just fixed this, or better yet had it working from release.

Off to Prusa I go!

zbrozek, 2025-04-05:
It's still impossible to pair a printer in LAN mode with the current release of BambuStudio. It works fine in Orcaslicer.

ManniAT, 2025-07-06:
I have the same issue with my H2D.
Had to use "Lan only", because cloud printing wasn't possible due to "service not reachable" and "printer offline" messages.
Sometimes it works - after restart. But max. once. After sending one printjob I get the FTP error every time I try to send another one.

skintigh, 2025-07-11:
Make sure our drivers are up to date, or keep an eye out for updates. It appears updating my Intel ethernet driver somehow solved my FTP/firewall issues. Before then I could only print if I turned off my firewall every time, even though the firewall logs showed nothing blocked. I still don't understand why that would cause/fix the issue.

papparaz, 2026-11-14:
Had the same problem with a project of my own. Solved it by changing the font of an embossing from 'Arial Bold' do 'Arial': I used the bold font oftrn in the past and never had a similar problem ??????