bambulab/BambuStudio #3576: Bambu studio won´t run, need help.

shanenoo, 2024-02-12:
### Bambu Studio Version

Bambu_Studio_win_public-v01.08.04.51-20240117164301

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 22631.3085

### Additional system information

 Processor: AMD Ryzen 7 5700U with Radeon Graphics 1.80 GHz
 Installed RAM: 16,0 GB (15,3 GB usable)
 Device ID: 75745A0C-47B7-47C3-B080-C3E0C16FFF2E
 Windows 11 pro, version 23h2

### Printer

Bambu X1 Carbon

### How to reproduce

I tried to run Bambu Studio but it doesn´t even launch at the beginning after installing it, I tried also to download and run OrcaSlicer but the problem is the same.

I tried also uninstalling the Windows Media Player, putting of the Firewall, re-installing the drivers, and also installing older versions of Bambu, but the problem is the same, I can´t run it.

### Actual results

It doesn´t launch, just the circle of charging the app goes on and nothing else happens.

### Expected results

It should run the app

### Project file & Debug log uploads

No file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-02-14:
@shanenoo 
could you help to try following things:
1. check whether there is a process named 'bambu-studio' in the task manager?
2. give a short video to record it
3. upload the log here(log can be found at C:\Users\user name\AppData\ BambuStudio\log)
4. remove directory C:\Users\user name\AppData\ BambuStudio and re-launch

shanenoo, 2024-02-16:
Hello, 

It doesn´t work still, I´ll attach the video.

https://github.com/bambulab/BambuStudio/assets/159763494/4948fdb7-2a3f-41b4-8b01-aa37a8ffbd11

lanewei120, 2024-02-17:
could you upload a dmp file in C:\Users\user name\AppData\ BambuStudio\log\ here?
it seems a crash found on your side

and are you using dual monitor?
if yes, could you also try to plug-out one monitor and re-launch again?

shanenoo, 2024-02-19:
I was using dual monitor and the problem was that seems like, when I unplugged the monitors it worked normal, now I connected them and it works, thank you for your help and time.

closed 2024-02-19 by shanenoo

walterwongbbl, 2024-02-19:
@shanenoo 
This should be a issue, if you could provide the dmp file, that would be great.

shanenoo, 2024-02-19:
[bambu-studio.exe.20568.dmp](https://github.com/bambulab/BambuStudio/files/14329918/bambu-studio.exe.20568.dmp)
You guys mean this?

lanewei120, 2024-02-26:
> [bambu-studio.exe.20568.dmp](https://github.com/bambulab/BambuStudio/files/14329918/bambu-studio.exe.20568.dmp) You guys mean this?

yes, thanks very much
sorry for the late reply

could you also help to upload file C:\Windows\System32\DriverStore\FileRepository\u0387389.inf_amd64_995be970e30b8c79\B385477\atio6axx.dll
here

we needs this dll to analysis the call stack

MischaI, 2024-09-14:
same there

lanewei120, 2024-09-16:
> same here: don´t start [bambu-studio.exe.12136.dmp](https://github.com/user-attachments/files/17003531/bambu-studio.exe.12136.dmp) [bambu-studio.exe(1).12136.dmp](https://github.com/user-attachments/files/17003532/bambu-studio.exe.1.12136.dmp)

Hello @MischaI 
have you tried these steps

> * check whether there is a process named 'bambu-studio' in the task manager?
> * give a short video to record it
> * upload the log here(log can be found at C:\Users\user name\AppData\ BambuStudio\log)
> * remove directory C:\Users\user name\AppData\ BambuStudio and re-launch

MischaI, 2024-09-17:
1. no prozess
3. all logs from
4. already did it, not help
[bambu-studio.exe(1).10056.dmp](https://github.com/user-attachments/files/17035193/bambu-studio.exe.1.10056.dmp)
[bambu-studio.exe.10056.dmp](https://github.com/user-attachments/files/17035194/bambu-studio.exe.10056.dmp)
[bambu-studio.exe(1).19204.dmp](https://github.com/user-attachments/files/17035195/bambu-studio.exe.1.19204.dmp)
[bambu-studio.exe.19204.dmp](https://github.com/user-attachments/files/17035196/bambu-studio.exe.19204.dmp)

lanewei120, 2024-09-18:
@MischaI 
are you also use dual monitor?
if yes, could you also try to plug-out one and restart for a try

is there any logs like this?
[image]
if yes, please upload these logs

MischaI, 2024-09-18:
Yes, I have 2 Monitors and Laptop. If i unplug usb docking station, it starts. But if i plug it back, i can´t start bambu studio.
And in Bambu Studio i can´t synchronize Filaments, after change to printer tab and i see my filaments, but back, i cant synchronise filaments.

MischaI, 2024-09-18:
after cleaning log from \AppData\Roaming\BambuStudio\log i havent eny file with .log.0
after unpluging docking station and successful start bambu studio i have .log.0 file
[debug_Wed_Sep_18_18_18_44_8220.log - Kopie.0.zip](https://github.com/user-attachments/files/17047101/debug_Wed_Sep_18_18_18_44_8220.log.-.Kopie.0.zip)

lanewei120, 2024-09-19:
thanks， we will try to reproduce it here

> after cleaning log from \AppData\Roaming\BambuStudio\log i havent eny file with .log.0 after unpluging docking station and successful start bambu studio i have .log.0 file [...]

MischaI, 2024-09-19:
> thanks， we will try to reproduce it here

log looks ok, but this log are created after i unplug docking station. i should  search such logs before i unplug docking station.
1. clear log directory
2. plug docking station in
3. try to start bambu studion
4. search, if .log.0 created. 

If not, what schould i do?

lanewei120, 2024-09-19:
> log looks ok, but this log are created after i unplug docking station. i should search such logs before i unplug docking station.
>
> 1. clear log directory
> 2. plug docking station in
> 3. try to start [...]

yes, just do as this, the log should appear

could you show us your hardware connections?
we can not reproduce it when using two monitors
let's check the hardware environment first

MischaI, 2024-09-19:
Laptop: Lenovo Legion 5 Pro 16 
Docking Station:  https://www.amazon.de/dp/B0BWSZCB4V?ref=ppx_yo2ov_dt_b_fed_asin_title
Monitor 1: https://www.amazon.de/dp/B083QQ6V1N?ref=ppx_yo2ov_dt_b_fed_asin_title
Monitor 2: old samsung 27 inch
both connected via hdmi

lanewei120, 2024-09-19:
> Laptop: Lenovo Legion 5 Pro 16 Docking Station: https://www.amazon.de/dp/B0BWSZCB4V?ref=ppx_yo2ov_dt_b_fed_asin_title Monitor 1: https://www.amazon.de/dp/B083QQ6V1N?ref=ppx_yo2ov_dt_b_fed_asin_title Monitor 2: old samsung 27 inch both connected via hdmi

so it seems related with docker station
could you send the log to us after the station plugged-in if exist?

fahaaaxiki, 2024-09-19:
> Laptop: Lenovo Legion 5 Pro 16 Docking Station: https://www.amazon.de/dp/B0BWSZCB4V?ref=ppx_yo2ov_dt_b_fed_asin_title Monitor 1: https://www.amazon.de/dp/B083QQ6V1N?ref=ppx_yo2ov_dt_b_fed_asin_title Monitor 2: old samsung 27 inch both connected via hdmi

Got it, I will simulate this hardware environment locally to try to reproduce the issue

MischaI, 2024-09-19:
> so it seems related with docker station could you send the log to us after the station plugged-in if exist?

nope, log file not exists. only dmp

MischaI, 2024-10-01:
Any solution?

bneres, 2026-02-14:
I'm having the exact same problem.

I reinstalled Windows 11 twice and Bambu Studio won't run.

I've already deleted my profile, tried running it as Administrator, and installed an older version. I can't work, the system isn't functioning.