bambulab/BambuStudio #3579: Ironing angle/Ironing direction is no longer visible

Yathu22, 2024-02-13:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 12.6.7

### Printer

Bambu P1S

### How to reproduce

Go to Quality, then Ironing.

### Actual results

Ironing angle/Ironing direction is not visible

### Expected results

we should see Ironing angle/Ironing direction

### Project file & Debug log uploads

Project file not required

### Checklist of files to include

- [ ] Log file
- [ ] Project file

LJKraus, 2024-02-15:
Try using Preferences (Ctrl+P), select "Develop Mode".

Previously reported as [Issue #3555](https://github.com/bambulab/BambuStudio/issues/3555) for Windows.

[Bambu wiki claims](https://wiki.bambulab.com/en/software/bambu-studio/release/release-note-1-8-2#:~:text=Support%20to%20set,smoother%20ironing%20results.) ironing angle feature was added in release 1.8.2, but it is only visible in develop mode.

XunZhangBambu, 2024-02-19:
@Yathu22  You can see the option in Develop Mode
[image]

closed 2024-02-28 by XunZhangBambu