bambulab/BambuStudio #3581: Color Mixing / Half-Tone

gallowayk, 2024-02-13:
**Describe the solution you'd like**
For those of us with AMS, I'd like to suggest "color mixing" by way of alternating colors between layers. Users should be able to achieve other color palettes with the combination of certain filament colors.

In the image below, you can see that the model and support the alternating red and yellow are flushing into is giving a rather nice orange color.
[image]

Ideally, the minimum color set for mixing users with a single AMS would use is CMYK or

White. Red, Blue, Yellow 

or 

Black, Red, Blue, Yellow

The expanded set would require two AMS units and would thus include: Black, White, Red, Yellow, Blue, any three other colors of the user's choosing.

Color combinations should be validated beforehand through practical use.

Examples of colors achievable:
Pink = Red + White
Sky Blue = Blue + White
Pastel Yellow = Yellow + White
Purple = Red + Blue
Green = Blue + Yellow
Etc.

Bambu Studio Behavior:
- Paint colors on parts as normal
- Bambu Studio determines if a painted color is achievable through half-tone (half layer height) mixing with colors currently loaded in the user's AMS.
- If the colors loaded in the AMS support it (i.e. pink), BS slices and creates the gcode to alternate between the two colors required.

TAGE7, 2025-07-06:
any update on this feature? with the H2D printer this should now be 100% a feature to make even more color possibilities!

gallowayk, 2025-12-12:
Even with my H2C, I think this could allow us to reduce the number of filament colors needed to be mixed in a bunch of cool ways for many shades.