bambulab/BambuStudio #3583: Support wall loops count is entirely ignored

RnDMonkey, 2024-02-14:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro 22H2

### Additional system information

Not performance or rendering issue.

### Printer

Bambu Lab P1S

### How to reproduce

1. Open attached project file
2. Slice plate
3. Preview layer 4
4. Note support interface pattern number of wall loops

### Actual results

Support interface pattern number of wall loops is 0:
[image]

### Expected results

Support interface pattern should have 2 wall loops. Lack of wall loops makes support layer less cohesive.
Cross-posted on Bambu Lab forum at https://forum.bambulab.com/t/support-wall-loops-count-is-entirely-ignored/56922

### Project file & Debug log uploads

[Basketball with pocket test.zip](https://github.com/bambulab/BambuStudio/files/14274301/Basketball.with.pocket.test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2024-02-18:
This option is originally designed for tree support only. We'll open it for normal support in the next version.

RnDMonkey, 2024-02-19:
Great, thank you!