bambulab/BambuStudio #3586: Start printer preparing when downloaing the job

radekkpl, 2024-02-14:
The printer is idle state when downloading print job.

I would like to see a speed improvement, that few lines of g-code to printer be sended at the begining of the print, so printer can load the filament from AMS, make bed calibration, nozzle cleaning even maybe pre first layer scan during downloading the job.

It can really speed up the time i need to spend at printer at the begining.

sodachen22, 2024-02-21:
@radekkpl This is a good idea. The difficulty seems to be that GCODE is packaged into 3MF. We have made some optimizations. For example, the A1 machine can start heating the hot bed when downloading the printing task. We may not be able to load the filament from AMS in advance because we don’t know what filament will be used until the print file is downloaded and parsed.

nhnifong, 2024-02-22:
Could bed levelling be started without the 3mf file?

radekkpl, 2024-02-27:
I am not sure about bed levelling, but load material from AMS is deffinitelly possible. Now before i click to send print i am manually clicking to AMS and loading the correct material. It really helps.

radekkpl, 2024-02-28:
[sodachen22](https://github.com/sodachen22)
Bambu studio should know when project is sliced the first filament used. So when clicking to send print, application can run same command like manual clicking to AMS for filament loading.