bambulab/BambuStudio #3589: I dont have the Colour Symbol anymore

ONLYTM, 2024-02-14:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

11

### Printer

P1S

### How to reproduce

cant use the Colour Option on BambuStudio

### Actual results

nothing

### Expected results

I need the Symbol to print in more than one colour

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Xboxer, 2024-02-14:
You need to add a second Filament to enable the Colour Icon

ONLYTM, 2024-02-15:
Thank you 🙏🏻Gesendet mit der mobilen Mail AppAm 14.02.24 um 23:10 schrieb Xboxer
                
            
            
                
                    Von: "Xboxer" ***@***.***>Datum: 14. Februar 2024An: "bambulab/BambuStudio" ***@***.***>Cc: "Author" ***@***.***>,"ONLYTM" ***@***.***>Betreff: Re: [bambulab/BambuStudio] I dont have the Colour Symbol anymore (Issue #3589)
                
                
You need to add a second Filament to enable the Colour Icon

lanewei120, 2024-02-18:
you can also find some guide on our wiki

https://wiki.bambulab.com/en/software/bambu-studio/color-painting-tool
https://wiki.bambulab.com/en/x1/manual/multi-color-printing

lanewei120, 2024-02-18:
as it is not a bug, close it currently

closed 2024-02-18 by lanewei120