bambulab/BambuStudio #3590: Thumbnail aspect ratio incorrect

CraigLampe, 2024-02-14:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sonoma 14.2.1

### Additional system information

Apple M1 Pro

### Printer

X1 Carbon

### How to reproduce

all models have thumnail in 3mf file.  Excellent feature!!  however they all look squished in Y direction (aspect ration is incorrect)

### Actual results

unable to fix

### Expected results

aspect ration should be accurate 1:1 pixel:pixel

### Project file & Debug log uploads

all files do this....

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-02-18:
Hi @CraigLampe 
thanks for reporting this

we use axonometric drawing to generate the thumbnail
maybe that is what you mean

could you give more pictures for the example

CraigLampe, 2024-02-18:
￼

All of these images are shorter than they should be on my MacPro MacOS
Similar on X1 Carbon built in screen
If stretched vertically they would look more REAL

CraigLampe, 2024-02-18:
￼
This image is much more realistic!  has good aspect ratio..

lanewei120, 2024-02-19:
@CraigLampe 
thanks for your good suggestion~

as the plate is 256*256
we had to keep the original size, maybe we can just extend the pictures in vertical with transparent content

closed 2024-08-08 by lanewei120