bambulab/BambuStudio #3591: Re-importing a non BL .3mf should not reset the per object settings already in the project

RealTadango, 2024-02-15:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Not a PC related issue

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Import a multi object file (.3mf in my case) and import it as a single model with multiple parts. Change wall loops for each part. Save the project and close studio. Save the .3mf file again from cad tooling (fusion 360). Open the project again in studio and you see the import questions again. After the import all changes made to the objects (filament, profile overrides) are reset to the default. 

### Actual results

All object settings are reset to default and there is no option to retain them or view them.

### Expected results

All objects settings should be kept if the amount of parts and names are the same. This is a big issue when you modify an object with complex per object settings in the project. You want to retain these settings and now they are lost. I need to keep all settings in notes because they cannot be retrieved from the project. When importing a non BL .3mf file settings should be as they are and not be reset.

### Project file & Debug log uploads

[outside-haaks.zip](https://github.com/bambulab/BambuStudio/files/14293436/outside-haaks.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

RealTadango, 2024-02-15:
It looks like when you import a .3mf file IT becomes the project. That is not correct. The .3mf should be used for importing the models only, like an STL.

RealTadango, 2024-02-15:
So basically, when importing a .3mf it should ask if you want to open the file as a project or only import the models from the file.

DanBao-Bambu, 2024-02-18:
I think asking users whether to import a 3mf file as a Project or as a model when loading can cause inconvenience for many users.
A possible approach is for Studio to provide an entry point for importing models when loading 3mf files. The file imported with this entry will not be imported as a Project, so a naming window will pop up when saving the modified file to distinguish it from the imported file and not overwrite the original file.

RealTadango, 2024-02-18:
But opening an STL does not do that. It saves a project with the same name as the STL file. Opening and importing should behave differently but how.... A question might be the clearer solution.