bambulab/BambuStudio #3594: Clicking one filament resets another

Kingsidorak, 2024-02-15:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

P1

### How to reproduce

1. Have at least 2 filaments set to select from (filament type doesn't matter)
2. Change the settings in one of them
3. Click on the second one, and it will ask if you want to transfer the settings, or discard the settings
4. Click Discard

### Actual results

Clicking discard to not have the settings transfer from filament 1 to filament 2 resets the settings back to default for filament 1

### Expected results

Filament 1 should not be affected by changes or discard of changes to filament 2, or any other filaments

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-02-18:
currently the text descriptions are not correct
[image]

we should not use transfer

DanBao-Bambu, 2024-02-21:
@Kingsidorak I understand your requirements, but currently there can only be one edited preset in Studio. If you switch to another preset and the previous changes are not saved, you can only choose to save or discard them.